FreeCAD/FreeCAD #19015: Sketch: Coincidence to an arc snaps to nearest point of the whole circle, even if is not part of the visible part of arc

elsni, 2025-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
* create new body
* create simple sketch with a rounded rectangle
* pad the object to a body
* create sketch on top surface
* project the rounded corners
* create line within the boundaries of the body
* use coincident tool on one endpoint of the line and the projected corner arc

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * InventorLoader 1.5.1
  * sheetmetal 0.6.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-12:
This is intended behavior. You should be able to drag the point.

https://wiki.freecad.org/Sketcher_ConstrainPointOnObject
https://wiki.freecad.org/Sketcher_ArcOverlay

elsni, 2025-01-12:
Is it really intentional that the point snaps to invisible geometry?  It sometimes can't be moved around when the line is already constraint. To be honest, it's a pain.

Roy-043, 2025-01-12:
Try moving the point closer to the visible portion of the arc before applying the constraint.

elsni, 2025-01-18:
Thanks, but this is only a workaround for (in my opinion - as a user) unexpected behavior. 

If I (as a user) constrain an endpoint of a line to an arc, I expect it to snap to the arc.

If I want the line to snap to the circle which the arc is part of, I would draw a construction circle with same diameter and center point as the arc and constrain to that. That would also make the sketch more readable.

elsni, 2025-01-18:
Thanks. I noticed that other CAD software behaves similar (which I find weird). So I have to reconsider my expectations. ;)

closed 2025-01-18 by elsni