FreeCAD/FreeCAD #19018: Part Attachment

nkrill46, 2025-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When following "Basic Attachment Tutorial" I find that the triangles attached to the sketch as instructed will not perform the "Pocket" function as shown, but produce an error message instead. For step by step analysis please see: https://wiki.freecad.org/Basic_Attachment_Tutorial#Map_Path_Parameter
[Part Attachment.zip](https://github.com/user-attachments/files/18390963/Part.Attachment.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39804 (Git) Conda
Build type: Release
Branch: main
Hash: 73f0e165bee7837704d374d0b1da430027e99fdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion
Installed mods: 
  * Assembly4 0.50.15
  * Curves 0.6.51
  * fasteners 0.5.31
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Anything else?

[Part Attachment.zip](https://github.com/user-attachments/files/18390963/Part.Attachment.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-12:
If you compare the images in the tutorial with your file you will notice that your IsoTri sketch is attach to a different corner of the rectangle. This is probably due to the way the rectangle was created. If you select "Through all" for the pocket and check the "Reversed" option things should work fine.

[Tree-Tutorial-rot.FCStd.zip](https://github.com/user-attachments/files/18391182/Tree-Tutorial-rot.FCStd.zip) (remove the .zip extension)

FEA-eng, 2025-09-27:
@nkrill46 or @Roy-043 Can it be closed?

Roy-043, 2025-09-27:
A paragraph has been added to the tutorial to address this.
https://wiki.freecad.org/index.php?title=Basic_Attachment_Tutorial&diff=1560083&oldid=1559626

closed 2025-09-27 by Roy-043