FreeCAD/FreeCAD #19022: Solus Plasma repo version of FreeCAD has graphics problem but AppImage works fine.

banger696-glitch, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening Freecad and creating a new draft document the drawing space is non responsive and nothing can be drawn but AppImage works fine.

### Full version info

```shell
OS: Solus 4.6 Convergence (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.10, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

AppImage
OS: Solus 4.6 Convergence (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

Safe mode has the same problem on Solus package version.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

banger696-glitch, 2025-01-13:
May be helpful seems to work fine with Nouveau driver but has problems in Wayland and Nvidia driver 565.77, ok for both in X11 desktop.

FreeCAD
FreeCAD 1.0.0, Libs: 1.0.0RUnknown
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Coin warning in cc_glglue_instance(): Error when setting up the GL context. This can happen if there is no current context, or if the context has been set up incorrectly.
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.

furgo16, 2025-01-13:
Looking at the about info of the local build, FreeCAD is running in Wayland, which is not yet supported. This might be a duplicate of https://github.com/FreeCAD/FreeCAD/issues/16393

furgo16, 2025-01-16:
@luzpaz @maxwxyz on the topic of marking issues as duplicates, perhaps you might find this interesting:

> To close an issue as a duplicate on GitHub, you can add a comment to the issue stating that it's a duplicate and referencing the original issue. For example, you can write "Duplicate of #[issue number]" in the comment, and GitHub will recognize it as a duplicate¹. This method helps to link related issues together and makes it clear to anyone viewing the issue that it has been closed because it is a duplicate, not because it was not planned or completed.

Source: https://github.com/orgs/community/discussions/78159#discussioncomment-7814934

But more interestingly:

> These new features are all available in public preview for you to try. To gain access for your organization, please sign up [here](https://github.com/features/issues/signup).
>
> You can now close an issue as a duplicate of another issue, making it easier to manage your issues and provide more clarity on why they were closed.
> 
> When closing an issue, select Close as duplicate from the dropdown to search for and select the duplicate issue. You’ll then see an event in the timeline and note at the top making it clear why it was closed.

Source: https://github.blog/changelog/2024-12-12-github-issues-projects-close-issue-as-a-duplicate-rest-api-for-sub-issues-and-more/

Do you think it might be a good idea to apply for the FreeCAD organization in GitHub to join the public preview program to have early access to new features such as Close as duplicate, sub-issues, issue types, advanced search?

maxwxyz, 2025-01-16:
@chennes @yorikvanhavre could you sign up as organization?

maxwxyz, 2025-01-16:
It now works with the new UI

closed 2025-01-16 by maxwxyz