FreeCAD/FreeCAD #19023: After starting freeCAD for the first time, there is no response when opening a file on the secondary monitor.

uyoufu, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After starting freeCAD for the first time, there is no response when opening a file on the secondary monitor.
When opening a file on the primary monitor, close the file and drag it to the secondary monitor, it works well.
Uninstalling freeCAD and deleting the files in Roaming cannot restore the normal state. It can only be restored after restarting the computer.
video bellow:

https://github.com/user-attachments/assets/7c1513d6-aedb-421a-8946-a13f0923dbe1

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-13:
Any chance you can test the weekly to reproduce? (And report back) https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

uyoufu, 2025-01-14:
There are still bugs.
I tested the latest version: FreeCAD_weekly-builds-39841-conda-Windows-x86_64-py311.7z. 
When started for the first time, the file still cannot be opened on the secondary monitor.
my windows version is: Windows 11, 24H2, os version: 26100.2605

maxwxyz, 2025-01-15:
is it always on the same screen (hardware) or depending on which monitor you set in Windows settings as main?

uyoufu, 2025-01-16:
> is it always on the same screen (hardware) or depending on which monitor you set in Windows settings as main?

it always only works on main minitor which main I was set

luzpaz, 2025-02-01:
we need windows testers to confirm

maxwxyz, 2025-02-03:
I cannot confirm, maybe hardware specific?
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.53
  * dodo 1.0.1
  * fasteners 0.5.35
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * kicadStepUpMod 11.5.2
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * parts_library
  * parts_library.backup1737145514.009932
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5

```

uyoufu, 2025-02-17:
Yes, it should be a hardware problem, my computer is configured as:
CPU: AMD R9 7950X
GPU: AMD Radeon(TM) Graphic
         Nvidia GTX 1080
Two external monitors are connected.

When I use an external monitor to my laptop, this problem does not occur on the secondary monitor

This bug is exceptionally hard to troubleshoot，anyway, thanks very much!

luzpaz, 2025-11-19:
Can you test on latest 1.2.0-dev (aka weekly) ?

uyoufu, 2025-11-24:
``` info
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251119 (Git shallow)
Build date: 2025/11/19 23:28:17
Build type: Release
Branch: main
Hash: deab0abf8e61d69f2208b794f64ac56428467fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Navigation Style/Orbit Style/Rotation Mode: Revit/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/166.066/1
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.60.2
  * geodata
  * Reinforcement
  * Road 2025.2.6
  * trails

```

Yes, I think it's Qt compatibility problem, I will ceate a simple Qt window to test

Lgt2x, 2025-12-24:
I cannot reproduce the bug with FreeCAD on Windows 11. I've tested both 1.0.2 binary and local master build using pixi. If it only affects a single user, maybe issue priority should not be "high"

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

uyoufu, 2026-09-03:
I reinstalled the system today, and after testing it on versions 1.0.2 and 1.1.3 respectively, I found that everything was normal.

I'm sorry this issue wasted everyone's time, and thank you very much for your efforts.

closed 2026-09-03 by uyoufu