FreeCAD/FreeCAD #19025: Styles: FreeCAD Light is missing left padding for workbench selector list items

furgo16, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

See screenshot above. When the workbench selector is set to be in <kbd>ComboBox</kbd> mode, and the list of workbenches is expanded from the menu, all list items are right next to the widget's left margin, where some sort of left padding would be expected instead.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Curves 0.6.40
  * BIM021 2021.12.0
  * Assembly4 0.50.14
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-14:
@MisterMakerNL FYI

furgo16, 2025-01-14:
In comparison, using the FreeCAD Classic theme, the left padding is there:

[image]

(_For some reason the workbench names are in bold, but I guess that's another issue_)

MisterMakerNL, 2025-01-14:
Need to know how if this element has an id and what it is. As far as I see it is currently not themed.

maxwxyz, 2025-01-15:
@furgo16 is it the same as with FreeCAD Dark?

MisterMakerNL, 2025-05-21:
This is fixed in https://github.com/FreeCAD/FreeCAD/pull/20528/commits/e127a3c317724f3d56f273bfebd5336de810f4d7#

closed 2025-07-11 by kadet1090