FreeCAD/FreeCAD #19026: PartDesign: Linear-pattern : Can't use different spacing

PaddleStroke, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
Check the above picture where you need to make a linear transform, but the spacing is not continuous.

There is currently no easy way of doing this.

On other CADs they do it like this : 

> On Catia the intervals can be set manually. You telle the feature to make N copies spaced by a given distance and it shows the N dimensions in the 3D view. You can edit the ones you need while keeping the default value for the other.

> Same in fusion360, there you have a checkbox by every instance in the 3d view

### Full version info

```shell
1.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pure-machinery, 2025-01-13:
I recommend a "Table" mode. Once selected you will have a table displayed for each instance where you can set the specific offset by adding an entry to the table. 

A trickier option would be pattern by point for this.

There may be other ways to approach this as well.

FlachyJoe, 2025-01-13:
Such behavior is available in the Lattice external workbench

[image]

tarman3, 2025-01-13:
**Lattice2** very powerful for complicated array

https://www.youtube.com/playlist?list=PLWuyJLVUNtc09gEkEg1WyqKDAE4MVkW-i

pierreporte, 2025-01-13:
Catia documentation for this: http://catiadoc.free.fr/online/prtug_C2/prtugbt0701.htm

Of course it works for circular patterns as well: http://catiadoc.free.fr/online/prtug_C2/prtugbt0702.htm

The problem with tables is that it lacks visual input. When the pattern count is high, finding the right cell can be harder that working visually. Even for simpler case it’s more user friendly. Though it would need 3D annotations that could be interacted with, see #13116 and https://github.com/FreeCAD/FreeCAD/issues/11214#issuecomment-2083512173.

redonwhite, 2025-01-13:
I have a PR adding a similar feature to Draft > PathArray:
https://github.com/FreeCAD/FreeCAD/pull/19017

[image]

pierreporte, 2025-02-05:
@maxwxyz  The issue has been solved for Draft but not PartDesign.