FreeCAD/FreeCAD #19027: Sketcher: Apply tangent between 2 horizontal lines gives redundant constraints.

PaddleStroke, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Apply a tangent between 2 horizontal or vertical or parallel lines gives redundant constraints.

Expected result : One of the horizontal is removed automatically to avoid redundant.

@maxwxyz I couldn't find, but perhaps there's already an open issue about this?

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

belinski-roman, 2025-01-13:
Maybe you should not delete it, but replace it with parallelism to another straight line?

PaddleStroke, 2025-03-06:
I cannot replicate this issue anymore. So someone must have fixed it since.

closed 2025-03-06 by PaddleStroke