FreeCAD/FreeCAD #19029: Some files open with the modified flag set

mpfusion, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some, but not all, files open with the modified flag set. Merely opening the file already sets the flag. Saving and re-opening the file still shows the modified flag.

[image]

An unmodified file should not show the modified flag. I'm not sure why only some but not all files are affected.
[open_modified.tar.gz](https://github.com/user-attachments/files/18395835/open_modified.tar.gz)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39806 (Git)
Build type: Release
Branch: main
Hash: 7829fcc4888bce067fe002a4d857a4953d2482fc
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.35
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-18:
I cannot reproduce with your example file. Is it the same in the stable 1.0 version for you?
Can you narrow it down which files have these issues? In you demo it has two views, 3D and a TD page, maybe it is only happening for files with multiple views?

theo-vt, 2025-09-24:
I believe it would, but that PR is kind of in a dead-end until I get around to fixing (and that people agree that it needs fixing) https://github.com/FreeCAD/FreeCAD/issues/23887