FreeCAD/FreeCAD #19030: Part Design: Pattern: Can't use another pattern

PaddleStroke, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
Trying to this. I expected that I would do a sketch of a single spike. Then do a pattern of it. Then a pattern of the pattern. But no that doesn't work.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pure-machinery, 2025-01-13:
I was able to do this with Multitransform (two linear patterns). 
[image]
[pattern.zip](https://github.com/user-attachments/files/18398451/pattern.zip)

```
OS: Arch Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * sheetmetal 0.6.12
  * fasteners 0.5.33
  * fcgear 1.0.0
  * OpenTheme 2024.9.1
  * IconThemes
```

Check if it's similar to what you want. (Of course this is only a workaround)

PaddleStroke, 2025-02-21:
@ziad-ashraf7 if you are interested in another small thing in part design, there's this issue that is a good candidate. And it is similar to the draft project you are finishing.