FreeCAD/FreeCAD #19031: FreeCAD 1.1.0dev appears to modify datum plane placements

furgo16, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I created a window assembly through several FreeCAD builds during the 0.22 dev cycle. I've kept working on it with FreeCAD 1.0.0. Model: [F4-B.zip](https://github.com/user-attachments/files/18397289/F4-B.zip)

However, when opening the file with FreeCAD 1.1.0dev recently, I've noticed the assembly now completely breaks after a recompute. Most of its components are rotated:

[image]

It's spectacular to watch it explode, but not fun considering the amount of time taken to create what should be a very simple assembly.

There are some errors in the report window, which appear TNP-related:

<details>

```
14:21:48  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed032.Reference2 missing element reference F4_B#Muntin_bar Muntin_bar.;#169:7b;:G#167;RFI;:H1e2:b,F;:H1e3,F;:H1e4:8,F.Face10
14:21:48  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed032.Reference2 missing element reference F4_B#Muntin_bar Muntin_bar.;#2801;:G#1c9a;CUT;:H1e1:c,V;:H1e3,V;:H1e4:8,V.Vertex19
14:21:48  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed030.Reference1 missing element reference F4_B#Muntin_bar001 Muntin_bar001.;#169:7b;:G#167;RFI;:H1e2:b,F;:H1e3,F;:H1e4:8,F.Face10
14:21:48  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed030.Reference1 missing element reference F4_B#Muntin_bar001 Muntin_bar001.;#2801;:G#1c9a;CUT;:H1e1:c,V;:H1e3,V;:H1e4:8,V.Vertex19
14:21:48  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed011.Reference1 missing element reference F4_B#Muntin_bar001 Muntin_bar001.;#e1;:H1e1,V;:H1e3,V;:H1e4:8,V.Vertex14
14:21:48  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed011.Reference1 missing element reference F4_B#Muntin_bar001 Muntin_bar001.;#e1;:H1e1,V;:H1e3,V;:H1e4:8,V.Vertex14
14:21:48  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed005.Reference1 missing element reference F4_B#Muntin_bar Muntin_bar.;#e1;:H1e1,V;:H1e3,V;:H1e4:8,V.Vertex14
14:21:48  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed005.Reference1 missing element reference F4_B#Muntin_bar Muntin_bar.;#e1;:H1e1,V;:H1e3,V;:H1e4:8,V.Vertex14
14:21:49  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed001.Reference2 missing element reference F4_B#Stile_outer_left Stile_outer_left.;#1e81;:G#6c4;CUT;:H-258:b,V;:H258,V.Vertex59
14:21:49  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed001.Reference2 missing element reference F4_B#Stile_outer_left Stile_outer_left.;#1e81;:G#6c4;CUT;:H-258:b,V;:H258,V.Vertex59
14:21:49  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Revolute.Reference1 missing element reference F4_B#Stile_outer_left Stile_outer_left.;#2928;:H212,V;:H258,V.Vertex5
14:21:49  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Revolute.Reference1 missing element reference F4_B#Stile_outer_left Stile_outer_left.;#2928;:H212,V;:H258,V.Vertex5
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed007.Reference2 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#30f3;:G#c6c;CUT;:H26a:b,V;:H27d,V.Vertex64
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed007.Reference2 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#30f3;:G#c6c;CUT;:H26a:b,V;:H27d,V.Vertex64
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed006.Reference2 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#2ae3;:G#c72;CUT;:H26a:b,V;:H27d,V.Vertex74
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed006.Reference2 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#2ae3;:G#c72;CUT;:H26a:b,V;:H27d,V.Vertex74
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed004.Reference2 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#30f5;:G#c78;CUT;:H26a:b,V;:H27d,V.Vertex84
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed004.Reference2 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#30f5;:G#c78;CUT;:H26a:b,V;:H27d,V.Vertex84
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed003.Reference2 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#c30:da;:G#c66;CUT;:H26a:b,V;:H27d,V.Vertex54
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed003.Reference2 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#c30:da;:G#c66;CUT;:H26a:b,V;:H27d,V.Vertex54
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed002.Reference1 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#30c5;:H26a,V;:H27d,V.Vertex6
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed002.Reference1 missing element reference F4_B#Stile_inner_left Stile_inner_left.;#30c5;:H26a,V;:H27d,V.Vertex6
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed007.Reference1 missing element reference F4_B#Muntin_single_left001 Muntin_single_left001.;#1019:105;:G#107b;CUT;:H-2ad:c,V;:H2ad,V.Vertex19
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed007.Reference1 missing element reference F4_B#Muntin_single_left001 Muntin_single_left001.;#1019:105;:G#107b;CUT;:H-2ad:c,V;:H2ad,V.Vertex19
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed006.Reference1 missing element reference F4_B#Muntin_single_left Muntin_single_left.;#1019:105;:G#107b;CUT;:H-2ad:c,V;:H2ad,V.Vertex19
14:21:51  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed006.Reference1 missing element reference F4_B#Muntin_single_left Muntin_single_left.;#1019:105;:G#107b;CUT;:H-2ad:c,V;:H2ad,V.Vertex19
14:21:52  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed005.Reference2 missing element reference F4_B#Muntin_partitioned_left Muntin_partitioned_left.;#1107;:H2cb,V;:H2d1,V.Vertex18
14:21:52  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed005.Reference2 missing element reference F4_B#Muntin_partitioned_left Muntin_partitioned_left.;#1107;:H2cb,V;:H2d1,V.Vertex18
14:21:52  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed004.Reference1 missing element reference F4_B#Muntin_partitioned_left Muntin_partitioned_left.;#10e8:10d;:G#1168;CUT;:H-2d1:c,V;:H2d1,V.Vertex23
14:21:52  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed004.Reference1 missing element reference F4_B#Muntin_partitioned_left Muntin_partitioned_left.;#10e8:10d;:G#1168;CUT;:H-2d1:c,V;:H2d1,V.Vertex23
14:21:53  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed002.Reference2 missing element reference F4_B#Rail_top_left Rail_top_left.;#3554;:H318,V;:H319,V.Vertex34
14:21:53  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed002.Reference2 missing element reference F4_B#Rail_top_left Rail_top_left.;#3554;:H318,V;:H319,V.Vertex34
14:21:54  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed003.Reference1 missing element reference F4_B#Rail_bottom_left Rail_bottom_left.;#1a86:174;:G#1b13;CUT;:H-32c:c,V;:H32c,V.Vertex7
14:21:54  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed003.Reference1 missing element reference F4_B#Rail_bottom_left Rail_bottom_left.;#1a86:174;:G#1b13;CUT;:H-32c:c,V;:H32c,V.Vertex7
14:21:54  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed012.Reference2 missing element reference F4_B#Stile_outer_right Stile_outer_right.;#341c;:M;MIR;:H214:7,V;:H22b,V;:H234:8,V.Vertex23
14:21:54  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed012.Reference2 missing element reference F4_B#Stile_outer_right Stile_outer_right.;#341c;:M;MIR;:H214:7,V;:H22b,V;:H234:8,V.Vertex23
14:21:54  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed010.Reference2 missing element reference F4_B#Stile_outer_right Stile_outer_right.;#3034;:M;MIR;:H214:7,V;:H22b,V;:H234:8,V.Vertex27
14:21:54  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed010.Reference2 missing element reference F4_B#Stile_outer_right Stile_outer_right.;#3034;:M;MIR;:H214:7,V;:H22b,V;:H234:8,V.Vertex27
14:21:54  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed008.Reference2 missing element reference F4_B#Stile_outer_right Stile_outer_right.;#361c;:M;MIR;:H214:7,V;:H22b,V;:H234:8,V.Vertex11
14:21:54  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed008.Reference2 missing element reference F4_B#Stile_outer_right Stile_outer_right.;#361c;:M;MIR;:H214:7,V;:H22b,V;:H234:8,V.Vertex11
14:21:54  <PropertyLinks> PropertyLinks.cpp(491): F4_B#Fixed014.Reference2 missing element reference F4_B#Stile_outer_right Stile_outer_right.;#782:a9;:M;MIR;:H214:7,V;:H22b,V;:H234:8,V.Vertex16
[... cut ...]
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39806 (Git) Snap 1262
Build type: Release
Branch: main
Hash: 7829fcc4888bce067fe002a4d857a4953d2482fc
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2025-01-13:
I believe the fix is already submitted https://github.com/FreeCAD/FreeCAD/pull/18704

furgo16, 2025-01-13:
Thank you! I'll test the branch to see if it fixes the issue on this model.

furgo16, 2025-01-13:
Pending further investigation, this would appear to be a duplicate of https://github.com/FreeCAD/FreeCAD/issues/18717 then

closed 2025-01-13 by maxwxyz