FreeCAD/FreeCAD #19047: Part: Placement of Part_CoordinateSystem is ignored when attaching to it

Roy-043, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Placement of the new Part_CoordinateSystem is ignored when a sketch is attached to it.

1. Open the attached [file](https://github.com/user-attachments/files/18401743/pd-attach-sketch-to-lcs-plane.FCStd.zip) (remove the .zip extension).
2. Activate the Body.
3. Select XY-plane001.
4. Invoke PartDesign_NewSketch.
5. result: the new sketch is aligned with the global XY plane.

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39804 (Git) Conda
Build type: Release
Branch: main
Hash: 73f0e165bee7837704d374d0b1da430027e99fdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-16:
@PaddleStroke FYI

PaddleStroke, 2025-01-17:
I can reproduce.
From what I see I would guess that the attachment engine is not taking into account the placement of the LCS. It's only using the placement of the plane. And the placement of XY-Plane001 is the same as XY-Plane.

The problem is actually that PartDesign_NewSketch is not setting up the AttachmentSupport correctly. It is setting `XY_Plane001 (XY-plane001)` when it should be setting `LCS [XZ_Plane001.]`
You can see that if you open the sketch's attachment dialog. And redo the selection of the plane there. Then it works correctly.

So it's the SketchWorkflow class that is not setting the attachment correctly.

Shkolik, 2025-02-06:
I see identical behavior not only on PD but in Part too. Placement of LCS ignored and redo selection doesn't work. So attachments to the LCS is useless now. Tried on builds 40006 and same behavior on 40077.

PaddleStroke, 2025-03-24:
So it appears that the problem comes from further, in SelectionFilter and Selection.
We do not have a resolve mode that enable us to get selection relative to an object (ie body in this case).

My suggestion is adding a function to SelectionSingleton (Selection.h) :

```
    /** Returns a vector of selection objects children of an object
     *
     * @param obj: Object within which you want to find a selection.
     * The selection objects returned will be direct children of Obj. The rest of the subname will be unresolved.
     * So this is equivalent to ResolveMode::NoResolve, but starting from obj.
     * For instance if you have : Assembly.Part.Body.LCS.PlaneXY
     * - If obj = Assembly : SelectionObject = Part, subname = "Body.LCS.PlaneXY."
     * - If obj = Part : SelectionObject = Body, subname = "LCS.PlaneXY."
     * - If obj = Body : SelectionObject = LCS, subname = "PlaneXY."
     * The docname used is the document of obj.
     * if obj is nullptr, this acts as getSelectionEx with empty docName
     * 
     * @param typeId: specify the type of object to be returned.
     * @param single: if set to true, then it will return an empty vector if
     * there is more than one selection.
     *
     * @return The returned vector reflects the sequence of selection.
     */
    std::vector<Gui::SelectionObject> getSelectionIn(App::DocumentObject* obj=nullptr,
            Base::Type typeId=App::DocumentObject::getClassTypeId(), bool single=false) const;
```

With this function we would be able to set PropertyXLinks correctly

closed 2025-04-07 by kadet1090