FreeCAD/FreeCAD #19051: Creating BIM project fails on MacOS (No module named 'typing_etensions')

risacher, 2025-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using weekly build from 2025-01-10 on MacOS, if I try to create a new Project from the BIM workbench's 3D/BIM menu, it fails with the below error.  I also tried this with a recent Linux build (on Ubuntu 24.04) and did not have the same problem.  This is maybe related to #18300, but that issue doesn't seem to have the same property of being MacOS-specific, and I can trigger it without opening an existing file.

```
14:33:28  Running the Python command 'BIM_Project' failed:
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/BIM/bimcommands/BimProject.py", line 49, in Activated
    from nativeifc import ifc_tools
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/BIM/nativeifc/ifc_tools.py", line 34, in <module>
    from ifcopenshell import geom
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/ifcopenshell/geom/__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/ifcopenshell/geom/occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

No module named 'typing_extensions'
```
I also note that `typing_extensions` looks like it is included, but I guess isn't being found for some reason: 
```
~$ find /Applications/FreeCAD.app -name 'typing_extensions*' -print
/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/pip/_vendor/typing_extensions.py
/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/setuptools/_vendor/typing_extensions-4.12.2.dist-info
/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/setuptools/_vendor/typing_extensions.py
```

### Full version info

```shell
OS: macOS 12.7.6
Architecture: x86_64
Version: 1.1.0dev.39804 (Git) Conda
Build type: Release
Branch: main
Hash: 73f0e165bee7837704d374d0b1da430027e99fdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * parts_library
  * Curves 0.6.53
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-01-14:
Duplicate of #18011

yorikvanhavre, 2025-01-14:
#18011 has been closed for another reason, but this problem is still there: Newer versions of ifcopenshell require python3-typing-extensions

@adrianinsaval is it doable to include that module in our builds?

luzpaz, 2025-01-14:
OP mentioned #18300, is there a relationship ?

yorikvanhavre, 2025-01-14:
> OP mentioned https://github.com/FreeCAD/FreeCAD/issues/18300, is there a relationship ?

I don't think so...

ErtugrulEge, 2025-01-14:
This is not just MacOS related. I am on windows and getting the very similar  messages always.For me creating a project doesn't produce the error automatically but **creating or deleting**  a wall ,line,window,door,column,beam... always produce 'typing_extensions' error. 

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39804 (Git) Conda
Build type: Release
Branch: main
Hash: 73f0e165bee7837704d374d0b1da430027e99fdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```
also I have a post on the forum https://forum.freecad.org/viewtopic.php?t=93277

```

21:53:23  pyException: Traceback (most recent call last):
  File "C:\Users\CCC\Desktop\FreeCAD-39804\Mod\BIM\nativeifc\ifc_observer.py", line 70, in slotDeletedObject
    from nativeifc import ifc_tools  # lazy loading
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\CCC\Desktop\FreeCAD-39804\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\CCC\Desktop\FreeCAD-39804\Mod\BIM\nativeifc\ifc_tools.py", line 34, in <module>
    from ifcopenshell import geom
  File "C:\Users\CCC\Desktop\FreeCAD-39804\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\CCC\Desktop\FreeCAD-39804\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "C:\Users\CCC\Desktop\FreeCAD-39804\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\CCC\Desktop\FreeCAD-39804\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
  File "C:\Users\CCC\Desktop\FreeCAD-39804\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'typing_extensions'
```

risacher, 2025-01-14:
FWIW. In the original report, I noted that the issue was observed on MacOS, but I could not replicate on Ubuntu.  Below is the Ubuntu setup that I'm using which seems not to exhibit the issue.  It's also a weekly build, but one week older.  I observe in passing that the Linux appimage includes Python 3.11.9, compared to 3.11.11 for the MacOS build.  (No idea if this is relevant.) 
```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

yorikvanhavre, 2025-01-15:
In any case, the typing_extension is a new module recently required by IfcOpenShell, I guess indeed as it goes spreading to the different platforms, we see this happening elsewhere too.

The problem here is that the typing_extensions is apparently well shipped with FreeCAD (one can check by typing: `import typing_extensions` in the Python console of FreeCAD), but for some reason IfcOpenShell can't find it?

ErtugrulEge, 2025-01-15:
import typing_extensions in the Python console also gives error

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * dxf-library
```

[image]

yorikvanhavre, 2025-01-16:
@adrianinsaval any idea of what's happening here?

adrianinsaval, 2025-01-16:
how recently was typing extensions added as dependency for ifcopenshell? AFAIK we are still shipping ifcopenshell 0.7 in the weeklies so I don't get what could have changed.

semhustej, 2025-01-18:
I no longer get `typing_extensions` or `curvestyles` errors in the latest weekly build.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

ErtugrulEge, 2025-01-18:
I just downloaded latest weekly and  drew a simple wall .Unfortunately the problem continues for me
```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

[image]

semhustej, 2025-01-18:
Correction: I tested in safe mode and the error still occurs on my machine too. The same as @ErtugrulEge

FEA-eng, 2025-02-04:
I also have this issue. The "No module named 'typing_extensions' error comes up when I try to create a new project in BIM or even open the built-in example.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/90.0633
```

yorikvanhavre, 2025-02-04:
> AFAIK we are still shipping ifcopenshell 0.7 in the weeklies so I don't get what could have changed.

You sure? The version in conda is now 0.8 IIRC

ErtugrulEge, 2025-02-09:
I have this message

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell ..0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/127.628
```

[image]

yorikvanhavre, 2025-02-10:
> IfcOpenShell ..0.0

This is weird and new :(

semhustej, 2025-02-10:
I can confirm the issue @ErtugrulEge reported:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

semhustej, 2025-02-10:
Also reported here with a workaround: https://forum.freecad.org/viewtopic.php?t=94644

yorikvanhavre, 2025-02-10:
hmm making people install typing_extensions themselves would indeed work, but I think it's best to have it included directly in our builds...

furgo16, 2025-02-10:
Strange that it's only an issue on macOS and Windows. At least on the snap package, we don't ship `typing_extensions` and I cannot reproduce the issue. Could it be that this was an issue fixed on ifcopenshell 0.8.0, but we're shipping an older version in the conda packages?

Looking at the reported ifcopenshell version in the about info (`0.0.0`. `..0.0`, `/.0.0`), it looks suspiciously like a bug that was fixed upstream with a less recent version of ifcopenshell. Which would mean conda is not providing the latest ifcopenshell. Just my guess.

furgo16, 2025-02-10:
This also seems related: https://github.com/FreeCAD/FreeCAD-Bundle/issues/314

psommerfeld, 2025-03-21:
The workaround mentioned [here](https://github.com/FreeCAD/FreeCAD/issues/19051#issuecomment-2647491577) is Windows-specific.
What is the MacOS equivalent?

jseyfert3, 2025-03-24:
I don't know if it's related or not, but I have been wanting to try to contribute to open-source projects I use for a while and just built FreeCAD from source. The build seemed to work fine, and FreeCAD ran. I opened the list of recommended issues for new contributors and looked at #20287. I tried to open the BIM Project Setup window and got:

```
19:26:03  IfcOpenShell was not found on this system. IFC support is disabled
19:26:03  Running the Python command 'BIM_Project' failed:
Traceback (most recent call last):
  File "/home/sasquatch/Software_Dev/freecad-source/build/Mod/BIM/bimcommands/BimProject.py", line 49, in Activated
    from nativeifc import ifc_tools
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sasquatch/Software_Dev/freecad-source/build/Mod/BIM/nativeifc/ifc_tools.py", line 52, in <module>
    raise e
  File "/home/sasquatch/Software_Dev/freecad-source/build/Mod/BIM/nativeifc/ifc_tools.py", line 35, in <module>
    import ifcopenshell
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

No module named 'ifcopenshell'
```

My own FreeCAD details:
```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40948 (Git)
Build date: 2025/03/23 21:49:13
Build type: Unknown
Branch: main
Hash: d5a1098add95e829ef8396736d23b043fbf9fa83
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/108.398/1
Installed mods: 
  * fasteners 0.5.37
  * freecad.gears 1.3.0
```

I then went and downloaded the FreeCAD_weekly-builds-40928-conda-Linux-x86_64-py311.AppImage and after adding qt6.conf and deleting qt.conf as mentioned [here](https://github.com/FreeCAD/FreeCAD-Bundle/issues/387) so it would open, I tried opening the BIM Project setup again and got the no module named typing_extensions error others have gotten:
```
19:53:06  Note: API not available due to missing dependencies: geometry.add_railing_representation - No module named 'typing_extensions'
19:53:06  Note: API not available due to missing dependencies: geometry.add_representation - No module named 'bpy'
19:53:06  Running the Python command 'BIM_Project' failed:
Traceback (most recent call last):
  File "/home/sasquatch/Software_Dev/squashfs-root/usr/Mod/BIM/bimcommands/BimProject.py", line 50, in Activated
    project = ifc_tools.create_document(FreeCAD.ActiveDocument)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sasquatch/Software_Dev/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 89, in create_document
    return create_document_object(document, filename, shapemode, strategy, silent)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sasquatch/Software_Dev/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 107, in create_document_object
    ifcfile, project, full = setup_project(obj, filename, shapemode, silent)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sasquatch/Software_Dev/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 176, in setup_project
    ifcfile = create_ifcfile()
              ^^^^^^^^^^^^^^^^
  File "/home/sasquatch/Software_Dev/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 195, in create_ifcfile
    ifcfile = api_run("project.create_file")
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sasquatch/Software_Dev/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 265, in api_run
    result = ifcopenshell.api.run(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sasquatch/Software_Dev/squashfs-root/usr/lib/python3.11/site-packages/ifcopenshell/api/__init__.py", line 88, in run
    importlib.import_module(f"ifcopenshell.api.{usecase_path}")
  File "/home/sasquatch/Software_Dev/squashfs-root/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<froze
No module named 'typing_extensions'
```

Details on this weekly build:
```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40928 (Git) Conda
Build date: 2025/03/22 22:27:10
Build type: Release
Branch: main
Hash: c15e45e7e2f5c071d3ac8a097194e417b561d7b4
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.398/1
Installed mods: 
  * fasteners 0.5.37
  * freecad.gears 1.3.0
```

risacher, 2025-05-05:
> The workaround mentioned [here](https://github.com/FreeCAD/FreeCAD/issues/19051#issuecomment-2647491577) is Windows-specific. What is the MacOS equivalent? 

Mac equivalent that worked for me: 
```
import subprocess;
subprocess.run(['/Applications/FreeCAD.app/Contents/Resources/bin/python', "-m", "pip", "install", "typing_extensions" ]);
```

markit, 2025-05-12:
Just to add a recent "GNU/Linux" side, AppImage FreeCAD_weekly-builds-41628-conda-Linux-x86_64-py311.AppImage
There is the "No module named 'typing_extensions'" but in the log you see also a "IfcOpenShell was not found on this system. IFC support is disabled" while "import ifcopenshell" works fine instead in the Python console.
```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git)
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/161.962/1

$ ./FreeCAD_weekly-builds-41628-conda-Linux-x86_64-py311.AppImage  --appimage-extract
$ cd squashfs-root/
$ ./AppRun 
FreeCAD 1.1.0, Libs: 1.1.0devR41628 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: File o directory non esistente
IfcOpenShell was not found on this system. IFC support is disabled
Traceback (most recent call last):
  File "/home/marco/AppImage/squashfs-root/usr/Mod/BIM/nativeifc/ifc_observer.py", line 135, in slotRemoveDynamicProperty
    from . import ifc_psets
  File "/home/marco/AppImage/squashfs-root/usr/Mod/BIM/nativeifc/ifc_psets.py", line 32, in <module>
    from . import ifc_tools
  File "/home/marco/AppImage/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 60, in <module>
    raise e
  File "/home/marco/AppImage/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 45, in <module>
    import ifcopenshell.geom
  File "/home/marco/AppImage/squashfs-root/usr/lib/python3.11/site-packages/ifcopenshell/geom/__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "/home/marco/AppImage/squashfs-root/usr/lib/python3.11/site-packages/ifcopenshell/geom/occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
<class 'ModuleNotFoundError'>: No module named 'typing_extensions'

$ find . -iname "*IfcOpenShell*"
./usr/lib/python3.11/site-packages/ifcopenshell
./usr/lib/python3.11/site-packages/ifcopenshell/__pycache__/ifcopenshell_wrapper.cpython-311.pyc
./usr/lib/python3.11/site-packages/ifcopenshell/ifcopenshell_wrapper.py
./usr/lib/python3.11/site-packages/ifcopenshell/_ifcopenshell_wrapper.cpython-311-x86_64-linux-gnu.so
$ 

Freecad, Python console:
Python 3.11.12 | packaged by conda-forge | (main, Apr 10 2025, 22:46:19) [GCC 13.3.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Start_Start',0)
>>> # Gui.runCommand('Std_Workbench',1)
>>> # Gui.runCommand('Std_ViewStatusBar',1)
>>> # Gui.runCommand('Std_Workbench',9)
>>> # Gui.runCommand('Std_Workbench',1)
>>> import PartDesignGui
>>> ### Begin command Std_New
>>> App.newDocument()
>>> # App.setActiveDocument("Unnamed")
>>> # App.ActiveDocument=App.getDocument("Unnamed")
>>> # Gui.ActiveDocument=Gui.getDocument("Unnamed")
>>> # Gui.activeDocument().activeView().viewDefaultOrientation()
>>> ### End command Std_New
>>> # Gui.runCommand('Std_OrthographicCamera',1)
>>> import ifcopenshell
>>> import typing_extensions
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ModuleNotFoundError: No module named 'typing_extensions'
>>> 

If I select "Strict IFC mode" icon on the bottom (default BIM Workbench)
$ ./AppRun 
FreeCAD 1.1.0, Libs: 1.1.0devR41628 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: File o directory non esistente
IfcOpenShell was not found on this system. IFC support is disabled
Traceback (most recent call last):
  File "/home/marco/AppImage/squashfs-root/usr/Mod/BIM/nativeifc/ifc_status.py", line 234, in on_toggle_lock
    lock_document()
  File "/home/marco/AppImage/squashfs-root/usr/Mod/BIM/nativeifc/ifc_status.py", line 358, in lock_document
    from . import ifc_tools  # lazy loading
    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/marco/AppImage/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 60, in <module>
    raise e
  File "/home/marco/AppImage/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 45, in <module>
    import ifcopenshell.geom
  File "/home/marco/AppImage/squashfs-root/usr/lib/python3.11/site-packages/ifcopenshell/geom/__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "/home/marco/AppImage/squashfs-root/usr/lib/python3.11/site-packages/ifcopenshell/geom/occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
ModuleNotFoundError: No module named 'typing_extensions'

```

zbynekwinkler, 2025-05-16:
I unpacked weekly AppImage as described above and installed `typing_extensions` into the unpacked environment and I can confirm the error is gone for me (on Linux).

yorikvanhavre, 2025-05-23:
I wasn't aware that typing_extensions was available via pip... That could be an easy solution here too, the BIM ifcopenshell command could then take care of that

zbynekwinkler, 2025-05-29:
> I wasn't aware that typing_extensions was available via pip... That could be an easy solution here too, the BIM ifcopenshell command could then take care of that

While trying to fix this issue, I've discovered that the AppImage is built using pixi which uses conda-forge as the source of the ifcopenshell package. The conda based package was missing typing_extensions dependency (see https://github.com/IfcOpenShell/IfcOpenShell/pull/6692 which contains the fix). So at some time in the future - when the updated version of ifcopenshell is published to conda forge - the problem could disappear. It was merged on May 11, so a release after that date could help. Currently the [last release is from May 15](https://anaconda.org/conda-forge/ifcopenshell).

A fix/workaround on the freecad side would involve running `pixi add typing_extensions` and commiting the resulting changes to the toml and lock files.

zeilsell-user1, 2025-06-02:
Is this normal for this community? A build issue that breaks the release for months and no one seems to be doing anything about it?

ErtugrulEge, 2025-06-03:
Now,I just downloaded latest version and I don't see any  `typing_extension`  message. I created some BIM and Draft objects and deleted theme everyting works fine.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

Richard-HeadCoach, 2025-06-03:
I downloaded 

https://github.com/FreeCAD/FreeCAD/releases/download/1.0.1/FreeCAD_1.0.1-conda-Linux-x86_64-py311.AppImage

On to an updated and upgraded Linux Mint laptop, and got the error when setting the dimensions for my building design to cm

zbynekwinkler, 2025-06-03:
I'd humbly suggest to wait for
- #21714

to be merged and a new weekly build was generated after that before further testing. 

Without it being merged, it is expected that the AppImage keeps throwing the error about `typing_extensions` missing.

closed 2025-06-05 by chennes

semhustej, 2025-06-07:
Tested with latest weekly and the issue is fixed. Thank you!

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

ErtugrulEge, 2025-06-07:
I have still the issue 

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38

```

[image]

FEA-eng, 2025-06-08:
I still have it too. Can the issue be reopened?

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

furgo16, 2025-06-08:
> OS: Windows 10 build 19045
> OS: Windows 11 build 26100

Note that this issue is for MacOS builds. If it's been fixed there, rather than reopen this issue I'd suggest creating a new one for the Windows builds.

ErtugrulEge, 2025-06-08:
> > OS: Windows 10 build 19045
> > OS: Windows 11 build 26100
> 
> Note that this issue is for MacOS builds. If it's been fixed there, rather than reopen this issue I'd suggest creating a new one for the Windows builds.

I opened the issue for windows

markit, 2025-06-09:
> > OS: Windows 10 build 19045
> > OS: Windows 11 build 26100
> 
> Note that this issue is for MacOS builds. If it's been fixed there, rather than reopen this issue I'd suggest creating a new one for the Windows builds.

This issue is a general one, in fact Windows and (me) GNU/Linux users contribute to this discussion so far.
The issue has not been solved, since is a module missing whaterver OS you create the build for (I can confirm for FreeCAD_weekly-builds-42095-conda-Linux-x86_64-py311.AppImage).
As far as I understand, this issue was far more complex (not difficult, but complex) involving the build system and automatic mechanisms that did not worked as expected.
I would prefer to reopen this instead of opening a new issue like @ErtugrulEge did, so a) we don't waste this discussion b) is chear that is the same issue

semhustej, 2025-06-10:
I tested Windows weekly build from both download locations. I tested also in Safe Mode. I am not able to reproduce the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1

```

```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1

```

But I was not able to reproduce the issue even with the 42095 build.

markit, 2025-06-10:
I can confirm that in GNU/Linux is (finally!) solved as well with weekly AppImage 
FreeCAD_weekly-builds-42133-conda-Linux-x86_64-py311.AppImage
Thanks a lot to all the people involved in the fix and testing of this annoying bug

Manfred-Gipp, 2025-06-19:
In FreeCAD_1.0.1-conda-Linux-x86_64-py311.AppImage this error still present

RacerDelux, 2025-06-20:
Just installed the latest version today on MacOS 15.5
I seem to have the same issue described above.

`22:38:48  (qt.qpa.fonts) Populating font family aliases took 122 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
22:38:48  Activating Ribbon Bar...
22:38:48  FreeCAD loaded without titlebar
22:38:48  Activating Save and restore...
22:38:50  FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon
22:42:14  pyException: Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/BIM/nativeifc/ifc_observer.py", line 85, in slotChangedDocument
    from nativeifc import ifc_tools  # lazy import
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/BIM/nativeifc/ifc_tools.py", line 36, in <module>
    from ifcopenshell import geom
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/ifcopenshell/geom/__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/ifcopenshell/geom/occ_utils.py", line 27, in <module>
    from typing_extensions import assert_never
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'typing_extensions'`

semhustej, 2025-06-20:
@RacerDelux can you please post the version of FreeCAD you are using?

RacerDelux, 2025-06-24:
> [@RacerDelux](https://github.com/RacerDelux) can you please post the version of FreeCAD you are using?

Apologies for the delay:
V: 1.0.1
Revision: 39285
Mac Arm64
Hash: 878f0b8

semhustej, 2025-06-25:
@RacerDelux taht makes sense. The issue is only fixed in 1.1 development version. 1.0.1 version you are using will not get this issue fixed.

RacerDelux, 2025-06-27:
> [@RacerDelux](https://github.com/RacerDelux) taht makes sense. The issue is only fixed in 1.1 development version. 1.0.1 version you are using will not get this issue fixed.

Ohh apologies, I thought that since the bug was closed it was released. Any idea when we might see the 1.1 version hit the wild?