FreeCAD/FreeCAD #19058: Core: Datums: No migration script with #18704

PaddleStroke, 2025-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After #18704, the change to the placement of datum axis has been undone.
Which means that files that were created these last 2 months are bad : 
[image]

A migration script should have been added to avoid this.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2025-01-14:
Given that https://github.com/FreeCAD/FreeCAD/pull/18332 was merged exactly a month ago (if I understand it correctly from the GitHub data), is adding a migration script worth the additional effort?

I assume the number of users affected will be low. Even if merged on the 13th of Dec, the feature would have been available only a few days later in the daily/weekly builds, which in turn are used only by a subset of FreeCAD users. There were Christmas and New Year holidays in between. More critical were the number of users affected by the issue fixed by https://github.com/FreeCAD/FreeCAD/pull/18704 (everyone who created an assembly with 0.22, 1.0 and 1.1 before the 13th of December). I [was one of them](https://github.com/FreeCAD/FreeCAD/issues/19031), and I'm thankful for both the feature and the fix!

Perhaps a middle ground, if doable, could be to simply create a macro attached to this issue, so that those affected can find it and run it once?

PaddleStroke, 2025-01-14:
You are correct, this may not be necessary. I said this because it's fairly simple to do, since the migration script was already writen to go from the inconsistent axis to the fixed axis. So going the other way would have been easy.

But yes that's maybe cluttering core with an unecessary migration script.

maxwxyz, 2025-01-14:
IMO also not needed, we explicitly mention that dev versions should not be used in a production environment and might break things.

PaddleStroke, 2025-01-14:
If anyone needs to migrate, here's how you can do it : 

add the following to LocalCoordinateSystem::migrateOriginPoint  : 

```
void LocalCoordinateSystem::migrateOriginPoint()
{
    // ... whatever in there

    for (auto* obj : features) {
        auto* feature = dynamic_cast <App::DatumElement*> (obj);
        if (!feature) { continue; }
        for (auto data : setupData) {
            // ensure the rotation is correct for the role
            if (std::strcmp(feature->Role.getValue(), data.role) == 0) {
                if (!feature->Placement.getValue().getRotation().isSame(data.rot)) {
                    feature->Placement.setValue(Base::Placement(Base::Vector3d(), data.rot));
                }
            }
        }
    }
}
```
Build. Then when you open a file it will migrate it automatically.