FreeCAD/FreeCAD #19060: Assembly: Cannot use datums axis anymore.

PaddleStroke, 2025-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The reversal of datum axis placement (#18704) broke the use of datums in assembly:

[image]

@wwmayer are you sure reversing this was the correct move? This inconsistent axis placement is a real long term technical debt. I think that keeping it rather than fixing the few things that broke is not the correct move.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2025-01-14:
> are you sure reversing this was the correct move?

Yes, this was the one and only correct fix. With your initial PR several features were left in a broken state (linear pattern, polar pattern and most of the FEM constraints). In the forum and the bug tracker there are at least a dozen of reported issues.

The most serious issue however is that old assembly projects are broken when opening and recomputing the document. Because if one of the axes is used as support then after the recompute its modified placement will be applied to the referencing object and this results into misplaced objects.

This is a general and not fixable problem. Even with the discussed general migration strategy it's not possible to correctly handle all use cases.

When comparing the placements of the three axes then the general migration strategy could map the old X axis to the new Z axis and the old Y axis to the new X axis in order to keep the placement. But a mapping from the old Z axis to another axis is not possible any more because the  placement of the new Y axis is quite different.

And this describes only one class of problems but we don't even know for sure what other problems the change may have caused in long-term.

> This inconsistent axis placement is a real long term technical debt.

This is not a technical debt but an implementation detail. The decision to use these placements for the axes was made more than 10 years ago and after that long time it's impossible to say what were the reasons for it. At least one thing is for sure: now it's too late to make any changes there. 

In the past there were ten or more places in the code base where this implementation detail was used to compute the expected direction vector. With my PR I have cleaned up the code by encapsulating the implementation details in some class methods. The same should be done for the assembly datum objects.

Now, what are the steps to reproduce the issue?

PaddleStroke, 2025-01-14:
I see.
Well the only issue (aside from this assembly one) that I see, is that if you do `datum.Placement` in python you get the wrong placement. Well at least not what the user would expect. But yes that's not so important, especially if the consequences of the migration were too big.

Anyhow, for this assembly bug just:
- create new file
- new assembly
- new body
- Show body origin
- Create fixed joint.
- Hover on the origin's Z axis. Result : the marker should overlap the origin. What happens : As in my picture, the marker is wrong.

PaddleStroke, 2025-01-14:
The problem comes from `findPlacement`, line 948 of UtilsAssembly.py
Currently we are returning `App.Placement()` for the case of datums as it was then correct. So here we should probably define a special case for Datum lines.

wwmayer, 2025-01-14:
> Hover on the origin's Z axis. Result : the marker should overlap the origin. What happens : As in my picture, the marker is wrong.

I cannot confirm a regression. For me the marker always lies in the XY plane independent of the hovered axis or plane. The exact same behaviour I get with a several weeks old version.

PaddleStroke, 2025-01-14:
@wwmayer Ah yes sorry there's a problem with the origin of Bodies.
The steps are in fact : 
- create new file
- new assembly
- new App::Part
- Show part origin
- Start fixed joint.
- Hover on the origin's Z axis. Result : the marker should overlap the origin. What happens : As in my picture, the marker is wrong.

The problem with the body is another one.

wwmayer, 2025-01-14:
The fix to this problem is to adjust `findPlacement` as you said. Add this block after the check for `obj` and before the calls of getElementName:
```
    if obj.TypeId == "App::Line":
        if obj.Role == "X_Axis":
            return App.Placement(App.Vector(), App.Rotation(0.5, 0.5, 0.5, 0.5))
        if obj.Role == "Y_Axis":
            return App.Placement(App.Vector(), App.Rotation(0.5, 0.5, 0.5, 0.5))
        if obj.Role == "Z_Axis":
            return App.Placement(App.Vector(), App.Rotation(-0.5, 0.5, -0.5, 0.5))
```

PaddleStroke, 2025-01-14:
Great. Note that is also bug with lines that don't have a role (see my initial post with a random line).

furgo16, 2025-01-14:
What is the practical implication of this issue for FreeCAD 1.1.0dev Assembly users?

I'm trying to decide whether I should stop using 1.1.0dev to work on existing assemblies and go back to 1.0.0 instead. I read the description, but I cannot quite figure it out.

PaddleStroke, 2025-01-15:
@furgo16 If you reated files after #18332 merged (about 1 month ago), then you have the problem that those files have been migrated to the axis placement fix. Which has been undone by #18704. The issue you are going to face is that those files are now migrated and there's no migration back. See for more details: 
https://github.com/FreeCAD/FreeCAD/issues/19058

Aside from this migration problem, if you have assemblies that use datums (but that was not supported before #18332 right?) just wait until there's a fix for this issue. And your files should be ok

furgo16, 2025-01-15:
Thanks for the detailed explanation. My assemblies were created a few months ago, on 0.22, and are regularly edited. My (Part Design) components use datums, but the assembly only uses fixed and revolute joints without any explicit reference to datums. So if I understand it correctly, I can keep using 1.1.0dev.

closed 2025-03-06 by chennes