FreeCAD/FreeCAD #19062: Problems with forum

tarman3, 2025-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are some problems with forum today or is it just me??
[image]

### Full version info

```shell
*
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-01-14:
Everything seems fine for me... Please try again?

tarman3, 2025-01-14:
Ok, probably the problem on my side
Thank you

yorikvanhavre, 2025-01-14:
Ok! Closing then!

FlachyJoe, 2025-01-14:
I'm facing some **502 Bad gateway** errors since about 12:00 UTC. The server is very long to respond and some attached picture are not displayed.

tarman3, 2025-01-14:
The same situation.
Now little bit better

tarman3, 2025-01-15:
Something's still wrong

Tried to check hosting by this services
- https://www.host-tracker.com
- https://http.app
sometimes forum.freecad.org was not response

It’s strange that site freecad.org the always answers

yorikvanhavre, 2025-01-15:
Indeed I can't access the forum either, now...
Probably our good old problem, AI bots clogging everything...
@kkremitzki could you have a look?

FlachyJoe, 2025-01-15:
A fix I seen for too insistent bots:
https://seenthis.net/messages/1091581

[EN translation]
> (for some reason I don't know, this beginning of the week, search engines are even more appalling than usual... I added a rule to fail2ban... SPIP blocks bots naturally, with an http 429 error when the load on the server is too high... but in fact, even the processing of this form of blocking is still too demanding... also, any IP address generating a 429 error ends up being blocked in the firewall, from now on... Internet bots have no respect for the resources they feed on)
> The late morning harvest is not impressive, but it is effective. 
```
Chain fail2ban-nginx-http-429 (1 references)
target prot opt ​​source destination
REJECT all -- 52.167.144.149 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 114.119.130.237 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 52.167.144.162 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 8.217.254.188 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 20.15.133.191 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 223.109.255.231 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 223.109.255.184 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 31.145.16.12 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 223.109.255.225 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 223.109.255.189 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 223.109.255.190 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 4.227.36.73 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 223.109.255.210 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 216.244.66.250 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 157.55.39.60 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 223.109.255.206 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 223.109.255.232 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 66.249.78.2 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 5.102.173.71 0.0.0.0/0 reject-with icmp-port-unreachable
REJECT all -- 66.249.69.103 0.0.0.0/0 reject-with icmp-port-unreachable
```

Roy-043, 2025-01-16:
As mentioned on the forum, the number of vistors is unusually high.
https://forum.freecad.org/viewtopic.php?p=804170#p804170

kaiwas, 2025-01-17:
At the beginning of this week I noticed slow work on my site. We looked at the logs. It turned out that **GPTBot** and **amazonbot** was making a lot of requests to the server.
When I banned it, the problems went away

And although the time interval is very similar, it is unlikely that the problem with the forum is related to this bot. But you could look at the logs.

Nowadays, artificial intelligence bots often visit sites. And they may overload them.

tarman3, 2025-01-23:
the problem continues

Syres916, 2025-02-05:
@kkremitzki I can well imagine new users to the forum being put off from using it as it's so slow, trying to use the `mention` tag has to be done manually as it doesn't give any suggestions due to the slow responsiveness, any assistance would be really appreciated, TIA.

Roy-043, 2025-02-05:
The number of users even went up to ca. 5300 today. This level of traffic seems suspicious.

tarman3, 2025-02-05:
https://www.youtube.com/watch?v=ZXsQAXx_ao0

semhustej, 2025-02-06:
I have not been able to access the forum in the past 12 hours.

tarman3, 2025-02-06:
Push 3..5 times to reload page and wait 1..5 minutes
Easy ))

yorikvanhavre, 2025-02-07:
The forum is fast again for me right now

tarman3, 2025-02-07:
No is only 2..3 hundreds of users online
Yesterday it was more than 7 thousands

tarman3, 2025-02-07:
as on a schedule, at exactly 16:00 another wave went

semhustej, 2025-02-07:
> as on a schedule some vile reptile pressed a button exactly at 16:00 another wave went

Yes - i am having problems accessing the forum right now.

edwilliams16, 2025-02-08:
Me too.  I did get on briefly, but couldn't do anything.  Only showed me and 30 guests.

kaiwas, 2025-02-08:
AI bots have something to read on the forum - that's why they attack.
Without setting up the server, the problem is unlikely to resolve itself.

Syres916, 2025-02-20:
The forum was stable and very quick to respond since the work carried out but unfortunately it's became slow this afternoon and now won't load for me at all.

kkremitzki, 2025-02-20:
Looking at it now

kkremitzki, 2025-02-20:
Marking as resolved - PHP-FPM issue, process tuning adjusted.

ErtugrulEge, 2025-02-24:
24 February 2025 13.02 Istanbul time forum  shows  504 Gateway Time-out error.

Syres916, 2025-02-27:
@kkremitzki I've been getting 502 Bad Gateway errors for the last hour

ErtugrulEge, 2025-03-04:
Forum is unreachable since morning hours 
[image]

semhustej, 2025-03-14:
Forum down for more than 1 hour:

[image]

kkremitzki, 2025-03-14:
I've further adjusted the PHP-FPM config to hopefully account for this. Please reopen/ping if it reappears.

semhustej, 2025-03-14:
> I've further adjusted the PHP-FPM config to hopefully account for this. Please reopen/ping if it reappears.

Seems to be working now, thank you!

Roy-043, 2025-03-14:
I am getting this now:
```
General Error
SQL ERROR [ mysqli ]

Too many connections [1040]

An sql error occurred while fetching this page. Please contact an administrator if this problem persists.
```

The number of users is bizarre: 29k+.

edwilliams16, 2025-03-14:
At 20:04 UTC getting 502/504 errors.  The forum has been up and down for the last hour.

BlueAG, 2025-03-14:
still down...!!

semhustej, 2025-03-15:
I can access the forum now: 
Sat Mar 15 10:15:50 2025 UTC

edwilliams16, 2025-03-15:
Here we go again. 502 at 10:38 PM saturday utc

xxxajk, 2025-03-16:
Still down Sun Mar 16 04:31:57 AM UTC 2025

semhustej, 2025-03-16:
I can confirm forum being down again.

Sun Mar 16 10:26:27 2025 UTC

kaiwas, 2025-03-16:
Doesn't anyone have access to configure nginx?

It's unlikely that this problem can be solved in other ways.

FlachyJoe, 2025-03-16:
A webmaster I know reports 400,000 blocked IP addresses in his database to curb AI hunger and prevent denial of service.

kkremitzki, 2025-03-16:
Should hopefully be up for good now!

semhustej, 2025-03-17:
> Should hopefully be up for good now!

Thank you!

Roy-043, 2025-03-17:
I think something is wrong with email confirmations. Since Friday afternoon no new user has activated their account.

edwilliams16, 2025-03-19:
502 again.  0330 UTC Wed March 19th

ErtugrulEge, 2025-03-19:
If it continues like this, Google algorithm will deindex.This is a serious problem for popularity of FC

semhustej, 2025-03-19:
Forum OK for me on
 Wed Mar 19 09:45:49 2025 UTC

jclxx, 2025-03-21:
Just noting that problem persists.  Forum says email sent, nothing arriving.  My email is at gmail.com.

kaiwas, 2025-03-22:
I don't understand what it says, but maybe it can be useful.

https://blog.cloudflare.com/ai-labyrinth/

jclxx, 2025-03-23:
I can confirm this appears functional now.

Activating "resend activation email" refreshed in immediately (0.10sec?), email arrived in under a second.

ErtugrulEge, 2025-03-24:
502 Bad Gateway

jclxx, 2025-03-24:
Currently `502 Bad Gateway`.

semhustej, 2025-03-24:
I am able to reach the forum now:
Mar 24 08:58:41 2025 UTC

FlachyJoe, 2025-03-26:
FYI: https://seenthis.net/messages/1105923

> We collected 470K IPv4s from a botnet that was trying to get all the content from our social network; it was behaving in such a way that we could track every single request it made. Since we blocked it, the server has been working much better; it hasn’t been running with such a low load for at least a year.
>  
> Can you try to use the ipset and tell us if it works ? Reply on seenthis or on mastodon, tagging [►](https://seenthis.net/sites/698828)[https://seenthis.net/people/biggrizzly](http://seenthis.net/people/biggrizzly) or [▻](https://seenthis.net/sites/7282461336008133946)https://framapiaf.org/@biggrizzly
> 
> Here is the latest file, more than 475K /32.
> 
> [►](https://seenthis.net/sites/7282461336008133936)https://www.partagerfichier.fr/download/2025-03-26-08-39-47_botnet-set.zip
> 
> It’s an ipset backup ([►](https://seenthis.net/sites/7282461336008129810)https://ipset.netfilter.org/ipset.man.html).
> 
> You can restore the file using this command line :
> 
> `ipset restore -file ./botnet-set`
> 
> You can declare it in iptables, using this command line :
> 
> `iptables -A INPUT -m set --match-set botnet-set src -j DROP`

semhustej, 2025-04-07:
> 502 Bad Gateway. 1640 UTC Mon April 7

I can confirm.

`Apr 07 17:04:11 2025 UTC`

ErtugrulEge, 2025-04-07:
Again the forum cannot be reached

semhustej, 2025-04-08:
I experience no problem with forum now.

`Tue Apr 08 15:39:23 2025 UTC`

edwilliams16, 2025-04-14:
8760 guests - forum very slow - almost unusable
Mon Apr 14 18:11:48 UTC 2025

fretlesskb, 2025-04-17:
I've been getting 502 bad gateway errors for a couple days now, I can't access any documentation or the forum

nigchr, 2025-04-26:
502 Bad Gateway since some days at https://wiki.freecad.org/

GMT: Saturday, 26. April 2025 13:28:37

luzpaz, 2025-05-05:
Should be solved thanks to @kkremitzki and the Anubis tool: https://blog.freecad.org/2025/04/30/forum-and-wiki-performance-anubis-deployed/

Closing this issue for now:

BTW,  there is a new separate issue:  
* #21122

closed 2025-05-05 by luzpaz