FreeCAD/FreeCAD #19069: Assembly: simulation crashes with segmentation fault

furgo16, 2025-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Download the [test file](https://github.com/user-attachments/files/18397289/F4-B.zip), uncompress it, and open it with FreeCAD
2. Double-click the `Window Assembly` component on the document tree to activate the assembly and load the Assembly workbench
3. Click on the `Create Simulation` command
4. On the Task panel, click on the plus sign button to add a motion. Leave the defaults and press `OK`
5. Click on `Generate`
6. FreeCAD crashes with the stack trace below:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7db47753d520]
#1  0x7db43046768d in MbD::EndFrameqc::ppAjOepEpE(unsigned long) from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0x24d
#2  0x7db43040befc in MbD::DirectionCosineIeqcJec::initializeGlobally() from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0x4c
#3  0x7db43040da74 in MbD::DirectionCosineIeqcJeqc::initializeGlobally() from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0x24
#4  0x7db430338215 in MbD::AngleZIecJec::initializeGlobally() from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0x15
#5  /snap/freecad/1264/usr/lib/libOndselSolver.so.1(+0x247a4f) [0x7db4303eda4f]
#6  0x7db4303ebf1f in MbD::ConstraintSet::constraintsDo(std::function<void (std::shared_ptr<MbD::Constraint>)> const&) from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0xaf
#7  0x7db4303ec0ef in MbD::ConstraintSet::initializeGlobally() from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0x4f
#8  /snap/freecad/1264/usr/lib/libOndselSolver.so.1(+0x38bf9f) [0x7db430531f9f]
#9  0x7db430534428 in std::function<void (std::shared_ptr<MbD::Item>)> std::for_each<__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::LimitIJ>*, std::vector<std::shared_ptr<MbD::LimitIJ>, std::allocator<std::shared_ptr<MbD::LimitIJ> > > >, std::function<void (std::shared_ptr<MbD::Item>)> >(__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::LimitIJ>*, std::vector<std::shared_ptr<MbD::LimitIJ>, std::allocator<std::shared_ptr<MbD::LimitIJ> > > >, __gnu_cxx::__normal_iterator<std::shared_ptr<MbD::LimitIJ>*, std::vector<std::shared_ptr<MbD::LimitIJ>, std::allocator<std::shared_ptr<MbD::LimitIJ> > > >, std::function<void (std::shared_ptr<MbD::Item>)>) from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0x58
#10  0x7db430530a35 in MbD::System::partsJointsMotionsLimitsForcesTorquesDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0x175
#11  0x7db430530c93 in MbD::System::initializeGlobally() from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0x53
#12  0x7db430530d88 in MbD::System::runKINEMATIC(std::shared_ptr<MbD::System>) from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0x98
#13  0x7db43034d504 in MbD::ASMTAssembly::runKINEMATIC() from /snap/freecad/1264/usr/lib/libOndselSolver.so.1+0xb4
#14  0x7db430f56b54 in Assembly::AssemblyObject::generateSimulation(App::DocumentObject*) from /snap/freecad/1264/usr/lib/AssemblyApp.so+0x244
#15  0x7db430f7c98e in Assembly::AssemblyObjectPy::generateSimulation(_object*) from /snap/freecad/1264/usr/lib/AssemblyApp.so+0x5e
#16  0x7db430f7c9fc in Assembly::AssemblyObjectPy::staticCallback_generateSimulation(_object*, _object*) from /snap/freecad/1264/usr/lib/AssemblyApp.so+0x2c
#17  /snap/freecad/1264/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128008) [0x7db478ee8008]
#18  /snap/freecad/1264/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7db478e9fadc]
#19  /snap/freecad/1264/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7db478e3ba1c]
#20  /snap/freecad/1264/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7db478f823af]
#21  /snap/freecad/1264/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0xe246c) [0x7db478ea246c]
#22  0x7db476946ec5 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /snap/freecad/1264/usr/lib/x86_64-linux-gnu/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15+0x95
#23  0x7db47694bc3d in PySide::SignalManager::qt_metacall(QObject*, QMetaObject::Call, int, void**) from /snap/freecad/1264/usr/lib/x86_64-linux-gnu/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15+0x14d
#24  /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f4666) [0x7db477c95666]
#25  0x7db4788ed9f6 in QAbstractButton::clicked(bool) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#26  /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x262c7e) [0x7db4788edc7e]
#27  /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2648a4) [0x7db4788ef8a4]
#28  0x7db4788efac7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#29  0x7db47883a94e in QWidget::event(QEvent*) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#30  0x7db4787f7763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#31  0x7db4787ff3a4 in QApplication::notify(QObject*, QEvent*) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#32  0x7db47a410d70 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1264/usr/lib/libFreeCADGui.so+0xd0
#33  0x7db477c5de4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#34  0x7db4787fde87 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#35  /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c91cb) [0x7db4788541cb]
#36  /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc525) [0x7db478857525]
#37  0x7db4787f7763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#38  0x7db47a410d70 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1264/usr/lib/libFreeCADGui.so+0xd0
#39  0x7db477c5de4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#40  0x7db4780861e7 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x687
#41  0x7db478059dbc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#42  /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5(+0x76f5e) [0x7db4716ddf5e]
#43  /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7db475736d3b]
#44  /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7db47578c258]
#45  /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7db4757343e3]
#46  0x7db477cb7338 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#47  0x7db477c5c76b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#48  0x7db477c64d14 in QCoreApplication::exec() from /snap/freecad/1264/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#49  0x7db47a34c7d4 in Gui::Application::runApplication() from /snap/freecad/1264/usr/lib/libFreeCADGui.so+0x6b4
#50  /snap/freecad/1264/usr/bin/FreeCAD(+0xa851) [0x635260de1851]
#51  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7db477524d90]
#52  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7db477524e40]
#53  /snap/freecad/1264/usr/bin/FreeCAD(+0xac35) [0x635260de1c35]
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Snap 1264
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2025-01-15:
@aiksiongkoh FYI

aiksiongkoh, 2025-01-15:
Looking into it.

furgo16, 2025-02-04:
@aiksiongkoh friendly reminder in case you have time to look into it. Thanks!

furgo16, 2025-07-17:
Attached ASMT file: 

[F4-B.asmt.txt](https://github.com/user-attachments/files/21303209/F4-B.asmt.txt)

Still reproduceable on:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42581 (Git) Snap 1576
Build date: 2025/07/16 02:43:57
Build type: Release
Branch: main
Hash: f2d0a69e015f528bba171c3d894445b0177f1b37
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Assembly4 0.50.17
  * Road 2025.2.22
  * Launcher
```