FreeCAD/FreeCAD #19070: [Feature Request] Add a Pose feature in the Assembly WB

X-Ryl669, 2025-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Related: #14786

Currently, as soon as an Assembly is activated, any move of part in the assembly is irreversible (except by Ctrl+Z). It would be very useful to have a "Pose" object that you can create that would take a snapshot of all parts' pose (position + rotation) in the assembly. Then the user could double click that object in the tree to move back all parts to the remembered pose.

This would be very useful because:
1. You can put the parts in the most challenging positions for checking tolerance or so (this usually involves long and tedious manipulations to get right) and don't be afraid to redo the operations. For example, you can have 2 poses for the extreme hinge position, so when editing, you can easily check that the new pad/pocket doesn't intersect other parts in that positions.
2. Revert easily to initial pose (the one used for editing the part, for example)
3. Show some specific important poses to someone without having to manipulate the assembly
4. Simplify animation (if you have a periodic pose, it's much easier to save and recall it)
 

### Full version info

```shell
1.1.dev
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oficinerobotica, 2025-01-15:
+1 for a "pose" functionality. In fusion360 is called "capture position" and is often used at the creation time of a component inside an assembly.  I find myself wishing for that often. 

The major benefit would be also when designing top down. Using the new "Insert new part" option in the Assembly workbench gives the possibility of referencing other components inside the assembly that might have moveable joints. Having a determined initial position of a component (part) ensures that whatever shape binder is used as reference will be back at it's initial position even if the assembly components move.

Sorry for my convoluted explanation , I hope I got my point across :)

mno13, 2025-01-19:
+1

Also, it could at the same time save visibility state for each component in assembly. Something like "Display configurations" in SE or "Scenes" in CATIA.

belinski-roman, 2025-01-19:
To do this, the suppression must work at the Body and Part levels. There is no such function now.

X-Ryl669, 2025-01-20:
???

belinski-roman, 2025-01-20:
> ???

If these are questions for me, what did I mean that now it is possible to suppress only individual operations inside the Body, but not the whole Body itself

X-Ryl669, 2025-04-08:
@sliptonic: Can you move this feature request to the project's [Integrated Assembly Workbench](https://github.com/orgs/FreeCAD/projects/7), please?

PaddleStroke, 2025-11-14:
I'll tackle this. It's indeed a useful tool we should have

pierreporte, 2025-11-16:
It would be great if that feature would also allow for visibility of parts and exploded views to be embedded.

luzpaz, 2025-11-17:
Maybe open a new ticket for that ?