FreeCAD/FreeCAD #19074: Change (replace) property prefix to property postfix

kaiwas, 2025-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the current implementation, parameters with group prefixes suffer from poor identification when called in expressions.

Despite the fact that the parameter is more important than the group, it is always displayed at the end.

1. I created three parameters with similar names in two groups.
2. I call the parameter I need by the group name (everything is called), or I call it by the property name, but I see the group name in the priorities.

[image]

My suggestion is to replace the prefix with a postfix. In this case, the parameter will be displayed with higher priority.
When called in expressions, the parameter will also become more important.

An approximate concept is shown in the image. (pic 3.4.)
[image]

Perhaps this change can break backward compatibility (I don't know). But if the "prefix code" is a pluggable cell, it may be possible to simply "replace" it to another place without compromising compatibility. You will just need to warn the user (with some phrase under this option).

Another option is to leave the prefix for those who find it convenient. And just add a postfix. (Something like a radio button. or a checkbox with a single choice)
Something like this
[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * addFC 1.0.6
  * freecad.gears 1.3.0
  * sheetmetal 0.6.13
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-01-15:
The current implementation follows the inheritance syntax `object.property` - from global to specific - and should not be break.

Yo can still hack the property name at your convenience by unchecking the prefix group name and append this name to your property.

kaiwas, 2025-01-15:
The group is simply a possibility of some cataloging. But we work with parameters.
despite the fact that the group is located "above" - ​​it is still less priority in calls (in expressions).

maxwxyz, 2025-01-15:
@FreeCAD/cad-advisory-group / @FreeCAD/design-working-group  any input?

FlachyJoe, 2025-01-15:
> from the programmers' point of view.

However you use this syntax when you refer to another object's property.

A possibility to clarify the groups usage could be to allow referring to their properties with a dot:
`object.group_property` could also be written `object.group.property`

kaiwas, 2025-01-16:
> A possibility to clarify the groups usage could be to allow referring to their properties with a dot:

Thank you! I know that I can add a group after the property name. But in this case I will have to do it every time for each new variable. If there are a lot of them, it is very labor-intensive. With the "postfix" checkbox, things would move much faster (if I understood your phrase correctly)

kaiwas, 2025-01-19:
Great suggestion on the forum.
https://forum.freecad.org/viewtopic.php?p=804274#p804274

johnsonm, 2025-12-02:
As a VarSet user, I found it very intuitive to have the group first. So the idea of switching from `_` to `.` rather than making it a suffix would make sense to me.