FreeCAD/FreeCAD #19077: Tool Controller not taken over when nothing is changed an ok is pressed in Cam Deburr Window

johandelange, 2025-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When the Tool controller is changed in the CAM deburr window en nothing else is changed the apply button does not enabled. Then when OK is pressed  and the the opereation opend again the old tool is back again.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Netherlands (en_NL)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * A2plus 0.4.67
  * A2plus.backup1725472595.2218428 0.4.60n
  * Assembly3 0.12.2
  * btl 0.9.9
  * Help.backup1725472601.345974 1.0.3
  * LCInterlocking
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-15:
could you share an example file as .zip and steps how to reproduce? TIA

johandelange, 2025-01-22:
Max,

First of all sorry for my late response.

I do not know  what I have to deliver a within zip file.

However I can give you the steps to reproduce the problem

First after started Freecad, select file new
In the part design  workbench add a new sketch (XY plane but I think that does not matter which plane) 
Draw a rectangle size (does not Mather)
Close the sketch
Pad to a  solid m(pad size does not matter)
Switch to the cam workbench en make a new cam job
Press oke
Add 2 or more tools to the Job with the toolkit library editor.  (>Camworkbenh ->CM-_toolbit libray editor)
Select Some of the upper lines of the cube to deburr
Select the deburr option (button  from the cam button bar)
Klik ok to select the default  tool or ( any tool you like)
The deburr window will open in the Task part of the screen.
Press ok to close the deburr window
Then Select in the operation part of the Cam job the newly added deburr operation and open the deburr window
Select ONLY a different tool 
 Notice that the apply button will not enable……
Press ok to close the deburr window
Reopen de the deburr operation check the tool an notice that the tool did not change

However if you open the deburr windows and change the tool AND something else, say the directon, the apply button will become enabled en when press ok the newly chosen tool will become active.

Hope this will help you to find the problem.

Thx Johan

maxwxyz, 2025-01-22:
It would be great if you could upload your FreeCAD example file after you have recreated the steps as a .zip directly in a comment on GitHub.
If possible, start recording a macro before recreating it and stop the recording afterwards. You can edit the macro and copy the text here.

johandelange, 2025-01-23:
I have Add a project and a macro file. I hope they will make the problem clear. Although the macro will not run on my installed freecad, because of some key problem.

[Deburrproblem.zip](https://github.com/user-attachments/files/18519561/Deburrproblem.zip)

closed 2025-09-02 by sliptonic