FreeCAD/FreeCAD #19080: CMake Error using FreeCAD1.0 with LibPack LibPack-1.0.0-v3.0.0-Release.7z

CheckTec, 2025-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can not create a error free cmake Project.

I am using
VirtualBox with Win10 pro
cmake 3.31.4
VisualStudio 2022, MSVC, version: 19.42.34436.0 (latest communiti version)
I Installed Qt6, PySide6 and Shiboken6 and GLS

I got a lot of errors on cmake.
What have I to do for a complet debugable installation of FreeCAD 1.0.

With FreeCAD 0,19 I had no such problems.

Attached you will find my log file from cmake
"C:\Program Files\CMake\bin\cmake" -Wno-dev -S D:\FreeCadSource\FreeCAD  -B D:\FreeCADCompiled  > MakeFreeCAD.log 2>&1

without the flag -Wno-dev  I got a much more errors.

### Full version info

```shell
No version dialog at this time
```

### Subproject(s) affected?

Project Tools & Websites

### Anything else?

[MakeFreeCAD.log](https://github.com/user-attachments/files/18426230/MakeFreeCAD.log)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-15:
@chennes FYI regarding LibPack

chennes, 2025-01-16:
You must run `git submodule update --init` to compile FreeCAD.

closed 2025-01-16 by maxwxyz