FreeCAD/FreeCAD #19081: Menu items not working on Mac Os version

andreaderoni, 2025-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Menu item **File/Edit Mode** cannot be found while menu item **File/Preferences** act like Edit Mode (Default/Transform/Cutting/Color sub menus). Have not found any workaround.

### Full version info

```shell
OS: macOS 12.7.6
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: ReleaseOS: macOS 12.7.6
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-15:
can you post some screenshots? What are you trying to do?

Syres916, 2025-01-15:
This looks remarkably similar to https://github.com/FreeCAD/FreeCAD/issues/10183 but there doesn't seem to have been an actual solution.

depthoffocus, 2025-01-15:
I can't reproduce this on 1.0 on intel Mac.

Preferences menu appears in the FreeCAD menu where it should be (on Mac) and works as normal.

[image]

Edit Mode is in the Edit menu -- is this not its normal location? -- and appears to work normally:

[image]

```
OS: macOS 15.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.35
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.53
```

@andreaderoni are you using FreeCAD in English or in a translation?

andreaderoni, 2025-01-16:
Soorry for makeing you wait, I've been out the whole day. The screenshot you put are just like what I get on my Linux versiion, not on my Mac one. I've tried the English versione and the one translated into Italien as well: same results. There iso no "Edit mode" under Edit Menu and, as I told, Preferences act like as Edit mode should.

[image]

closed 2025-01-16 by andreaderoni

andreaderoni, 2025-01-16:
Sorry for makeing you wait. I tried the english version and everythig seems ok. So we have a bug in the italian translated version, as you can see in te screenshot BUT only on Mac; Italian Linux version is all right

[image]