FreeCAD/FreeCAD #19085: Version: 1.1.0dev.39841 (Git) Conda AppImage_TECH DRAW_Projection_Group

APShark, 2025-01-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

OS: Debian GNU/Linux 12 (bookworm) (labwc:wlroots/LXDE-pi-labwc/xcb)
Architecture: arm64
Version: 1.1.0dev.39841 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * lattice2 1.0.0
  * freecad.gears 1.3.0
  * Curves 0.6.53
  * A2plus 0.4.68
  * Gridfinity 0.6.0
  * fasteners 0.5.34
  * OpenTheme 2024.9.1
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
  * ExplodedAssembly
  * BIM021 2021.12.0

When using the the TECH DRAW WORKBENCH (Version: 1.1.0dev.39841) to draw and dimension an existing part, the projections appear off my page. I am able to move all the projections back on the sheet, but main projection is locked an unmovable. I then tried to do the same thing in FreeCad 1.0 on the same computer, and I had no issues.
  

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (labwc:wlroots/LXDE-pi-labwc/xcb)
Architecture: arm64
Version: 1.1.0dev.39841 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * lattice2 1.0.0
  * freecad.gears 1.3.0
  * Curves 0.6.53
  * A2plus 0.4.68
  * Gridfinity 0.6.0
  * fasteners 0.5.34
  * OpenTheme 2024.9.1
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
  * ExplodedAssembly
  * BIM021 2021.12.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2025-01-16:
I experienced this issue a few days ago, using the daily builds (1.1.0dev) on Linux, via the snap package. Unfortunately, I did not take note of the steps to reproduce. As I was in a hurry and needed to get things done, I just switched to my 1.0.0 installation and created the TechDraw page from there and didn't bother debugging the issue on 1.1.0dev.

The symptoms were the same:
- The projection view was created off page
- I tried to move it by dragging the primary projection of the projection group as usual. It wouldn't move
- The secondary projections did react to dragging and could be moved. But that does not help if the projection group they're in cannot be moved
- It did not work with 1.1.0dev but it did work with 1.0.0

WandererFan, 2025-01-16:
i'm working on this one.   If you close/hide the page, then open/show it the projection group will sort itself out.

johndoe2323, 2025-12-29:
@luzpaz : can you pls check / set correct status? thank you!

WandererFan, 2025-12-30:
Did I miss a step?

[image]

OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.2.0dev.44818 +3 (Git)
Build date: 2025/12/29 14:46:41
Build type: Unknown
Branch: CosmeticEdgeDeleteCrashInScript
Hash: b2eb40e3c43deacd432b12ccb2b3536e22f4a0b3
Python 3.10.12, Qt 6.9.3, Coin 4.0.0, Vtk 7.1.1, boost 1_74, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.8.0
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/118.229/1.19792
Installed mods: 
  * ThreadProfile 1.98.0
  * freecad.gears 1.3.0
  * A2plus 0.4.68
  * sheetmetal 0.7.58
  * fasteners 0.5.43

johndoe2323, 2025-12-30:
I use the command from the menu to insert the projection group - please see below.

[image]

luzpaz, 2025-12-30:
I've reopened per request of @johndoe2323

maxwxyz, 2026-08-15:
Closed then if fixed.

commit 01f4e08f (2025-01-27): [TD]fix front view position in projection group (fix #19085)
closed 2026-08-15 by maxwxyz