FreeCAD/FreeCAD #19086: Make the difference between control and information properties more obvious.

kaiwas, 2025-01-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, there are some properties that only display information and do not control the geometry of the model (marked with blue arrows). Control properties are marked with red arrows.
This mainly concerns the BIM workbench, but may be elsewhere.
[image]

As we can see, visually, the properties look the same. There is no way to identify what type of properties these are (Control or Information). To understand the type of property, we have only one option - click on this property. If it is informational - the text color will be different from the text color in the control property.
[image]
[image]

This problem can be solved in different ways. The simplest of them is to make the text color "gray" in the properties panel itself.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * addFC 1.0.8
  * freecad.gears 1.3.0
  * sheetmetal 0.7.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2025-01-16:
This is something that, as a user who is still learning, I regularly stumble upon. It's not only for the BIM workbench, but other more visible properties exhibit this behavior too. One example is the Placement/Attachment Offset combo. If an object is not attached to another object, the Placement properties can be edited. Once it's attached, the Placement properties are read-only and only the Attachment Offset properties can be edited instead. While it makes sense once one understands how Attachment works, if there were some sort of immediate visual indication of the writeability of properties some of the confusion to new users would be alleviated.

> The simplest of them is to make the text color "gray" in the properties panel itself.

I like the simplicity of this proposal.

maxwxyz, 2025-01-16:
@FreeCAD/design-working-group are you ok with the proposal?

pierreporte, 2025-01-16:
The proposal makes sense.

MisterMakerNL, 2025-01-17:
Just a gray text won't work always, because like description if there is no text you still don't knowif itcan be filled in.
I'd suggest do it like op second screenshot and change the background color too plain white if it's editable.
Also we should try to use default disabled or non editable text styles as much as possible.
In de stylesheets an input field already changes if it can or cannot be used.

kaiwas, 2025-01-17:
> Just a gray text won't work always, because like "description" if there is no text you still don't know if it can be filled in.

Yes. This is an important note. It will be something for the design group to think about.
Perhaps the control fields should always be white, and the information fields should always be gray.
But probably in different styles the design can give different results.
There was an idea with a small "info" icon (similar to the "fx" expressions), but it probably won't look very good.

MisterMakerNL, 2025-01-17:
There is a design for editable text fields and non-editable no need to re-design it unless it's not clear enough. It just needs to be applied to the properties!

kaiwas, 2025-01-17:
> There is a design for editable text fields and non-editable no need to re-design it unless it's not clear enough. It just needs to be applied to the properties!

Excellent! So the task shouldn't be difficult.

MisterMakerNL, 2025-01-18:
This pretty big bug, and I'll try to fix this, I already looked into it but it's not aa simple as I hoped it too be. @Syres916 you got any tips/ideas?

maxwxyz, 2025-05-26:
should be fixed in #21571 @alfrix ?