FreeCAD/FreeCAD #19092: CAM Profile: Sub: Depths

HeinzLp, 2025-01-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the Depth Step the default Value is the OpToolDiameter as Default. 

### Full version info

```shell
In the Depth Step the default Value is the OpToolDiameter as Default.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-16:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- Test if the issue is still present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
- add full [About](https://wiki.freecad.org/About) info to the issue (`Help > About FreeCAD > Copy to clipboard`).
- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.

JohnOCFII, 2025-01-16:
This question was answered recently (September 2024)  by the primary CAM WorkBench maintainer.  The choice of `OpToolDiameter` as the default is as intended.  Here is the detailed answer (as I brought up the same question):

> It was chosen because we needed to have something as a default.  A hard-coded value is almost always wrong.  It made more sense to make it relational to the tool geometry.  The majority of early CAM users were word-workers using routers and a depth-of-cut based directly on tool diameter made sense.
> 
> A much smarter default value is possible if we consider the stock material but we don't want to force the user to do 50 things before they get a usable result.
> 
> In the meantime, a more reasonable default that works for wood and isn't horrible for aluminum is 30-50% of the tool diameter.  But that will still be completely wrong for titanium and stainless.

closed 2025-01-16 by maxwxyz

HeinzLp, 2025-01-17:
Thank you for your answer. I can understand that you sometimes use default values.

There are also a lot of other points to look at, power of the machine in all respects and how massive or fine the part will become, we try to build. 

 

Would it be a good idea to set a depth’s step factor for different materials, e.g. 0.4 for Aluminium, as a multiplication factor for the default value of the tool diameter?

 

Kind regards

Heinz