FreeCAD/FreeCAD #19093: Optional addon in packagexml throws an error when installing a workbench

APEbbers, 2025-01-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As a test the FreeCAD Ribbon has the following line in package.xml:
[image]  

This throws an error when people installing the ribbon. It can be ignored, but it won't install the addon
[image]

It is probally a feature in progress, but I mentioned it the give feedback on this.

### Full version info

```shell
Test with FreeCAD Ribbon
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-18:
Please add full about info, which version are you using?

APEbbers, 2025-01-18:
Hi,

about info is a bit difficult because the user that installs the FreeCAD Ribbon, gets the message. That can be any version or OS.
I believe the line to add an addon as optional in the package manager, is a new feature which is not working yet. 

The add-on manager should install the optional add-on as well. For FreeCAD Ribbon that is the SearchBar addon.
It is not an python package as the message suggests.

Also good to know, I'm the developer of the FreeCAD UI. Lupaz has asked to add this line to the package.xml. 
Some users reported this issue to me. That's why i've created this ticked. I can remove the line if that's better?

Below are the details of my FreeCAD version. However, this differs per user. 
OS: Windows 11 build 22635
Architecture: x86_64
Version: 1.1.0dev.39802 (Git)
Build type: Release
Branch: main
Hash: 558eb19103518090d494fcde0e8deabba830f734
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Netherlands (en_NL)
Stylesheet/Theme/QtStyle: Dracula.qss/FreeCAD Dark/
Installed mods: 
  * A2plus 0.4.68 (Disabled)
  * Assembly3 0.12.2 (Disabled)
  * Assembly4 0.50.15 (Disabled)
  * Behave-Dark-Colors 0.1.1
  * BillOfMaterials 1.0.2
  * boltsfc 2022.11.5
  * CfdOF 1.29.4
  * Curves 0.6.53
  * Dracula 0.0.5
  * DynamicData 2.70.0
  * ExtremeProDark 2.9.0
  * fasteners 0.5.33
  * FEMbyGEN 2.3.0 (Disabled)
  * FEM_FrontISTR 0.2.0
  * frame 0.1.1
  * FreeCAD-Ribbon 1.6.2.3
  * freecad.gears 1.2.0
  * IconThemes (Disabled)
  * Launcher
  * NikraDAP 2.0.0
  * OpenTheme 2024.9.1
  * PieMenu 1.11.2
  * Plot 2024.11.26
  * ProDarkThemePreferencePack 1.0.0
  * Render 2024.12.15
  * SearchBar 1.3.2
  * sheetmetal 0.7.3
  * StemfieWB 0.2.0
  * WorkFeature-WB

luzpaz, 2025-01-18:
@APEbbers In `package.xml` the depend tag references  `SearchBar Workbench`? What if you change it to `SearchBar` ? 

The same way it shows up in the About info: 

> SearchBar 1.3.2

APEbbers, 2025-01-18:
Hi, yes that was it. However now there is an another issue:

[image]

I've now tested it on my virtual linux enviroment because I'm working on an update currently.

luzpaz, 2025-01-18:
Does `/use/bin/python3` exist in your system?

APEbbers, 2025-01-18:
yes It does exists. But shouldn't the add-on manager use the python version that is shipped with FreeCAD? Otherwise, users must install python them self.

If you want to use pip from the python console in FreeCAD, you have to import pip first. that might be related?

APEbbers, 2025-01-19:
Sure the link is: https://github.com/APEbbers/FreeCAD-Ribbon/issues/237#issuecomment-2600840946.

I also updated the text on top

luzpaz, 2025-01-19:
> If you want to use pip from the python console in FreeCAD, you have to import pip first. that might be related?

@chennes is the solution to pre-load `pip` ?

chennes, 2025-01-19:
The selection of the Python that gets used is not in the Addon manager, that's now a global FreeCAD method, here:
https://github.com/FreeCAD/FreeCAD/blob/bf422367636a01d66fbfef68a8ddba0efea28880/src/Ext/freecad/utils.py#L32