FreeCAD/FreeCAD #19095: Part Design: missing "Create a datum" command

furgo16, 2025-01-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Up until FreeCAD 1.0.0, the Part Design Workbench had the ability to create datum objects (e.g. `PartDesign::Plane`) from its menu, as described at https://wiki.freecad.org/PartDesign_Plane and shown on this screenshot:

[image]
_Part Design Datum planes can still be added with FreeCAD 1.0.0_

However, in FreeCAD 1.1.0dev those menu entries are no longer available.

In case the change is intended, and the motivation is to favor the new `Part::DatumPlane` and associated datum objects, please consider:

1. **Consistency**: removing also the ability to create PartDesign datum objects from the tasks panel (see screenshot from FreeCAD 1.1.0dev). The user can inadvertently add and mix different object types depending on whether they try to add an e.g. plane from the commands toolbar(`Part::DatumPlane`) or the Tasks panel (`PartDesign::Plane`)
   [image]. 
3. **Feature parity**: bring `Part::DatumPlane` to feature parity with `PartDesign::Plane`. E.g. the new `Part::DatumPlane` is missing Width and Height properties (I can file a separate issue about that).

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39862 (Git) Snap 1266
Build type: Release
Branch: main
Hash: 201eca4d02b5969a2eeaefd871804854b9273867
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-16:
Where I can find this **Helper tool** menu ???

maxwxyz, 2025-01-17:
Width and height is deprecated (plane is indefinitely).
@PaddleStroke it seems the old commands are still in the task dialog.

furgo16, 2025-01-17:
> Width and height is deprecated (plane is indefinitely).

Oh, I did not know that. I use height and width all the time on planes and then I can use the planes as external geometry as a reference for sketches.

What will be the recommended workflow once planes can no longer be dimensioned? Attach a master sketch in construction mode to every plane and then reference that sketch? It seems to defeat the purpose of using planes as a reference.

maxwxyz, 2025-01-18:
IMO planes should not be used as workaround for geometry.

closed 2025-01-18 by maxwxyz