FreeCAD/FreeCAD #19098: Part Design Task panel does not show available operations/tasks when shape/sub-shape binders are selected.

neoh4x0r, 2025-01-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I tested this on commit 17985b45c6f5a4f256fc6c4d8aa9642d1030ba07 (v1.1.0 R39866)

Currently in the Part Design Workbench/Task Panel when a shape or sub-shape binder is selected in an active body operations like (pad, etc) are not shown, but the pad can be performed by selecting the appropriate option from the toolbar or part design menu.

This behavior can be confusing because it leads a user (eg. myself) into thinking that there are no available options for the selected binders (eg. you cannot pad it directly and must create a sketch and use the binder as external geometry in order to create the pad). 

### Full version info

```shell
FreeCAD 1.1.0 Revision: 39866 (Git), commit 17985b45c6f5a4f256fc6c4d8aa9642d1030ba07
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-17:
Can you please add the full [About](https://wiki.freecad.org/About) info to the original ticket? TIA

neoh4x0r, 2025-01-17:
> Can you please add the full [About](https://wiki.freecad.org/About) info to the original ticket? TIA

I just now added the information  from the about dialog.

I also updated the commit that was tested, the original one was wrong because it was related to my local checkout (where I committed some patches to ignore some files) -- that actual commit was 3385ac8a744632d722d59d7cb708fc0b02a70197.