FreeCAD/FreeCAD #19099: Copying to the clipboard property from a panel for use in expressions

kaiwas, 2025-01-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

At the moment we can insert a parameter into an expression only using "search" in the expression input window and tooltips. This is good, but when there are a lot of properties and similar objects, this causes difficulties, since it is necessary to constantly analyze which object corresponds to the required property

[image]

I propose to add the ability to copy a specific property to the clipboard in the taskbar. In the right-click menu when clicking on a property, add an item with copying, which contains the text of the expression with a regular name and with an internal name.
Approximate concept in the picture.

[image]

Such a change will allow you not to remember the names of properties in sets of objects and will significantly reduce the mental load on the user.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * addFC 1.0.8
  * freecad.gears 1.3.0
  * sheetmetal 0.7.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-18:
Sounds like a `Driver` in Blender

kaiwas, 2025-01-18:
There may be problems in the case when the expression length is very large.
Then it would probably be better to make a fixed length for the drop-down list with an unchangeable title.
but when we hover over the desired item, we get a pop-up hint with the full text of the expression.
The concept is in the picture.

(I think this will require the thoughts of the design group)

[image]