FreeCAD/FreeCAD #19100: Assembly: flexible assemblies cannot be deleted: `No document object found with name ''`

furgo16, 2025-01-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is a follow-up to #17903

Steps to reproduce:

1. Uncompress the [test file](https://github.com/user-attachments/files/17783800/test-exploded-view-sub-new_rigid.zip) and open it with FreeCAD
2. Double click on the `Assembly` object on the document tree to activate the assembly
3. Click on the Insert component command (`Assembly_InsertLink`) on the toolbar
4. On the Tasks panel dialog, Ensure `Rigid sub-assemblies` is unchecked
5. On the Tasks panel dialog click on the `Assembly001` object to insert it
6. **Notice**: the inserted object does not show a transform gizmo to move it around, as it generally happens with other inserted components
7. On the Tasks panel dialog, press `Ok`
8. Press `Ctrl+R` to recompute
9. **Notice**: the following warnings appears on the Report view
   ```test_exploded_view_sub_new_rigid#Fixed004 is unconnected to a grounded part so it is ignored.
      test_exploded_view_sub_new_rigid#Fixed004 is unconnected to a grounded part so it is ignored.
      1.327e-06 <App> Document.cpp(2903): test_exploded_view_sub_new_rigid#Fixed004 still touched after recompute
   ``` 
10. Select the `Assembly002` object in the document tree
11. Right-click on it to activate the context menu
12. Choose the `Delete` command from the context menu
13. **Expected**: the assembly is deleted
14. **Actual**: the assembly cannot be deleted. Instead, an error message is shown:
   ```
   11:49:54  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: No document object found with name ''
```

https://github.com/user-attachments/assets/f1a549e0-902b-4df7-8f8a-d25f4576db3a

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39868 (Git)
Build type: Release
Branch: main
Hash: a66da9b3d2281c5ef0a20ce14bacaf21f66f31f3
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * WikiPropertiesGenerator.FCMacro
  * fasteners 0.5.30
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-18:
@PaddleStroke FYI

PaddleStroke, 2025-07-18:
First thing, with the enclosed file, following the steps give me those errors messages directly : 

```
17:11:30  pyException: Traceback (most recent call last):
  File "C:\Users\Sky_l\Desktop\Freecad-Build\Mod\Assembly\JointObject.py", line 630, in execute
    and (joint.Reference1[1][0].find("?") != -1)
         ~~~~~~~~~~~~~~~~~~~^^^
<class 'IndexError'>: list index out of range
17:11:30  pyException: Traceback (most recent call last):
  File "C:\Users\Sky_l\Desktop\Freecad-Build\Mod\Assembly\JointObject.py", line 637, in execute
    and (joint.Reference2[1][0].find("?") != -1)
         ~~~~~~~~~~~~~~~~~~~^^^
```

The subassembly joint shows an error. After recompute the object is not in error anymore but then I get
```
17:15:59  test_exploded_view_sub_new_rigid#Fixed004 is unconnected to a grounded part so it is ignored.
17:15:59  test_exploded_view_sub_new_rigid#Fixed004 is unconnected to a grounded part so it is ignored.
17:15:59  1e-07 <App> Document.cpp(3039): test_exploded_view_sub_new_rigid#Fixed004 still touched after recompute
```