FreeCAD/FreeCAD #19101: BIM: Window tool button and Door tool button reversed

jag45, 2025-01-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the BIM workbench pressing the Window toolbar button inserts a door and pressing the Door toolbar button inserts a window.
This happens both in Version 1.0.0 and in the latest weekly 39841.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: MY-Darker-Flat.qss/MY-Darker-Flat/Fusion
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * FreeCAD-themes 2024.7.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-17:
@semhustej would you mind confirming ?

kaiwas, 2025-01-17:
It all depends on the preset you choose. At the moment, the window or door tool essentially does the same thing. Maybe someday Yorick will separate these two tools so that the tool matches the presets.

`Version: 1.1.0dev.39841 (Git) Conda`

https://github.com/user-attachments/assets/53efdca2-7d4e-4c80-99d0-a32bd48ac8fd

Roy-043, 2025-01-17:
@kaiwas is right. The door tool initially uses a door preset, but the user can choose any other preset, which is then stored for the next use of the tool. The same applies to the window tool.

paullee0, 2025-01-29:
So, nothing needs to be done, right?  Closing this issue?

luzpaz, 2025-01-29:
>  Maybe someday Yorik will separate these two tools so that the tool matches the presets.

Maybe that's the quintessential  ticket instead of the current one ?

semhustej, 2025-01-29:
I think the buttons are not reversed as written in the original post. It can be a bit confusing, but they are not switched from my testing.

What is confusing to me is that some of the parameters/settings of Arch_Window and BIM_Door tools are independent of each other, and some synchronize across these two tools. I think it would be nicer, if all of the settings were independent. I don't see a scenario, where I want to change door and window parameters to be the same. 

[image]

Yorik has refactoring of the Door/Window tool in plan, so I am not sure if working on the current tool makes sense (not sure when the new tool will be introduced).

But this is another issue, so the current issue (OP) can be closed I think.

yorikvanhavre, 2025-02-04:
Let's leave this open, as it is indeed confusing. What I think should happen:

- all parameters should be independent between door and window (even if they use the same default values)
- the list of presets should be separated in two, and each tool showing a different subset

I'm indeed working on a refactor of the tool, so I'll assign this to myself :)