FreeCAD/FreeCAD #19102: Quick measure: Error when selecting a circular edge

maxwxyz, 2025-01-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When selecting any circular edge in the 3D view, an error is show:
```Measurement::radius - Invalid References3D Provided```
I think this is related to the quick measure feature in the toolbar, as it happens on every selection and the radius is displayed as 0.

[image]

Demo:
[Unnamed.zip](https://github.com/user-attachments/files/18455718/Unnamed.zip)

Steps to reproduce:
1. Open example file.
2. Select any circular edge in the 3D view.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Measurement

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-17:
@PaddleStroke FYI

PaddleStroke, 2025-01-17:
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39845 +25 (Git)
Build type: Release
Branch: s02
Hash: de81fa3e805b8eba4d3e55dc80b67363f784ed02
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```
No error on my side

closed 2025-01-17 by maxwxyz