FreeCAD/FreeCAD #19104: Sketcher: units (mm) not showing properly in 'live' dimensions

semhustej, 2025-01-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- use Standard unit system and two decimal places (default settings)
- create new sketch
- create rectangle
- click first point
- move mouse far from the first point (ideally less than 10000mm and more than 1000mm)
- mm is not shown properly in the 'live' dimensions field

[image]

This has no effect on functionality as when one starts typing into the field, mm show up correctly.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Sketcher

### Anything else?

I could reproduce this in FC 1.0.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-17:
It is probably hard to get the size of these input boxes just right. It seems to unnecessary right margin/padding.

hyarion, 2025-01-18:
Input boxes should grow/shrink with the content inside.
A similar issue could be seen when renaming entries in model tree in overlay mode, that issue was fixed last month though.

semhustej, 2025-01-18:
> It is probably hard to get the size of these input boxes just right. There seems to be an unnecessary right margin/padding.

I understand. This is not a big deal, but something to keep in mind if it could be fixed easily. If it's not an easy fix, the issue can be closed.

It could be a bit confusing for new BIM users, who use larger values than mechanical users. With the default units setup, this behavior occurs. With some values, the unit can even look like meters instead of milimeters:

[image]