FreeCAD/FreeCAD #19107: [File System] On Linux, it is not possible in versions 1.1 to access files on other hard drives.

luisDias89, 2025-01-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the latest version of Freecad, access to the "/Media" folder—which provides access to disks, pen drives, etc. in the system—has stopped working in version 1.1.
Captura de ecrã de 2025-01-17 18-50-06

How to reproduce: Place a file on a USB drive or external disk, open version 1.1, and attempt to open the file.

In version 1.0 works:
Captura de ecrã de 2025-01-17 18-51-13

### Full version info

```shell
OS: Ubuntu Core 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39868 (Git) Snap 1268
Build type: Release
Branch: main
Hash: a66da9b3d2281c5ef0a20ce14bacaf21f66f31f3
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-01-17:
@luisDias89, could you apply the version info of working _FreeCAD_?

bdieterm, 2025-01-17:
For me this looks like a problem with restrictions from the snap sandbox, like issue #19008.

furgo16, 2025-01-17:
Another duplicate of https://github.com/FreeCAD/FreeCAD-snap/issues/58

This should fix it for you:

```
sudo snap connect freecad:removable-media
```

While at it, you can also try to set the `Dialog/DontUseNativeDialog` parameter in FreeCAD to `false` to use your distro's native file picker dialog. As described on https://wiki.freecad.org/Fine-tuning

luisDias89, 2025-01-18:
I thought it was due to the version, but when I compile FreeCAD on my PC, it remains the same.
[image]
OS: Freedesktop SDK 24.08 (Flatpak runtime) (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39868 (Git) Conda
Build type: Release
Branch: main
Hash: a66da9b3d2281c5ef0a20ce14bacaf21f66f31f3
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * Render 2024.12.15
  * OpenTheme 2024.9.1

luisDias89, 2025-01-18:
You're right, this OpenFileDialog window is a thousand times better. Why doesn't it come set up like this by default?

[image]

[image]

closed 2025-01-18 by maxwxyz