FreeCAD/FreeCAD #19109: Can only scale parameterized model by small amounts or it gets garbled

Lobanz, 2025-01-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

New to FreeCAD, but I'm a developer too and this dont seem right. I have a model that scales itself by referencing constraints between sketches.  Nothing crazy.  In the model, I change the diameter of the main object and the rest of the model is supposed to scale appropriately.  Works great as long as I dont scale too much (like over a factor of 2).  I have scaled the model in small increments from .25mm to 5000mm and it works fine as long as I do it in small steps.  If I go from, say, 100mm to 500mm, the part will get garbled.

I attached a zip of the FreeCAD file. If has a little spread sheet to change the outer diameter of the part for testing.

Maybe I'm doing somethiung wrong.  Anyway, hope that helps.

### Full version info

```shell
FreeCAD 1.0.0 Windows.
No add-ons.
Does the same thing in Safe Mode.
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]

[ShowerheadPattern.zip](https://github.com/user-attachments/files/18461624/ShowerheadPattern.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-18:
[image]

maxwxyz, 2025-01-18:
I also cannot reproduce this.
Could you try if you have the same issue in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
And please add the full about info.

tarman3, 2025-01-18:
My gif above demonstrate the problem
Just change dimension from 500 to 100
After that circles in Sketch changed places

[image]

I simplified the file
[ShowerheadPattern_.zip](https://github.com/user-attachments/files/18463723/ShowerheadPattern_.zip)

maxwxyz, 2025-01-18:
can confirm. Seems to mess up the constraints / expression.

Lobanz, 2025-01-18:
updated about info.  thx.  you guys rock.