FreeCAD/FreeCAD #19115: App.newDocument append "1" to given name.

PaddleStroke, 2025-01-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

`App.setActiveDocument("testFile")`

Creates a file named `testFile1`

'App.setActiveDocument("AnotherTest1")'

Creates a file named `AnotherTest2`

### Full version info

```shell
1.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kpemartin, 2025-01-19:
I'm unable to reproduce this. Here is what I get in the FreeCAD Python console when I try (with the auto-generated comments about active document changes etc removed):
```
>>> doc=App.newDocument("testFile")
>>> doc.Name
'testFile'
>>> doc.Label
'testFile'
>>> doc2=App.newDocument("AnotherTest1")
>>> doc2.Name
'AnotherTest1'
>>> docA=App.newDocument("testFile")
>>> docA.Name
'testFile1'
>>> doc2A=App.newDocument("AnotherTest1")
>>> doc2A.Name
'AnotherTest2'
```
Note that if you already have an open document with the Name `testFile` then a call to `App.newDocument("testFile")` will create a document with a name made unique by adding digits. Similarly, object Names are made unique within each document, and under most circumstances object Labels are made unique within each document as well.

I'll admit that it is a bit confusing that there is the method `Application.getUniqueDocumentName` to create a unique Document name, but that it is entirely unnecessary because `newDocument` makes the name unique anyway.

It would be clearer if either one of the following were true:
1. Names passed to `Application::newDocument` had to be different from the names of any existing documents, and it was the caller's responsibility to call `Application::getUniqueDocumentName` if the name already exists (`Application::getDocument(name)` returns a non-null object).
2. There were no `Application::getUniqueDocumentName` method at all, and it was expected that `Application::newDocument` would generate a unique name as required, and the name of that document could be retrieved by using its `Name` property.

I personally would prefer number 2, as number 1 is burdensome on any code creating a document, but currently we have a blend of numbers 1 and 2, and a lot of code (particularly unit tests, it seems) use something between the two models (regarding both Document and DocumentObject naming).

Perhaps the parameter name and documentation comments of `Application::newDocument` (and similarly `Document::addObject`) should point out that the name passed is only a suggested name, and that the actual object may have a different name. Documentation for the Python wrapper code would have to reflect this as well.

Pending any detailed steps to reproduce, this seems to be working by design.

kpemartin, 2025-01-20:
I don't know what's different, but I can only partly confirm:

> 
\>>> App.newDocument("testFile")
<Document object at 000001F12D8CD020>
\>>> # App.setActiveDocument("testFile")
\>>> # App.ActiveDocument=App.getDocument("testFile")
\>>> # Gui.ActiveDocument=Gui.getDocument("testFile")
\>>> # Gui.runCommand('Std_OrthographicCamera',1)
App.newDocument("AnotherTest1")
<Document object at 000001F12D4D9A50>
\>>> # App.setActiveDocument("AnotherTest1")
\>>> # App.ActiveDocument=App.getDocument("AnotherTest1")
\>>> # Gui.ActiveDocument=Gui.getDocument("AnotherTest1")

The document tree shows:
[image]

>
\>>> # with the second document active
\>>> App.activeDocument().Name
'AnotherTest1'
\>>> App.activeDocument().Label
'AnotherTest2'
\>>> # activating the first document using the mouse
\>>> # App.setActiveDocument("testFile")
\>>> # App.ActiveDocument=App.getDocument("testFile")
\>>> # Gui.ActiveDocument=Gui.getDocument("testFile")
\>>> App.activeDocument().Name
'testFile'
\>>> App.activeDocument().Label
'testFile'

It seems the documents, in my case, get the expected names "testFile" and "AnotherTest1" but the second document's Label is "AnotherTest2" which is what shows up in the document tree.

>
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39935 (Git)
Build type: Release
Branch: master
Hash: 49b304ae558553f9acd5872ccd30673bbd60bd19 (this morning's master branch.)
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/

At least this gives me something to look at in the debugger.

kpemartin, 2025-01-20:
My previous (2 days ago) inability to reproduce this may have been because I already had the new unique-name code.

Here's what I see happening:
If the call to newDocument is given a Name but no Label (the function argument is confusingly called userName), then the Name is used as a proposed Label. Then `Base::Tools::getUniqueName(userName, names);` (where `names` is a collection of all the Names—not Labels—of existing documents) is evaluated and the result used as the Label. Because the given userName ends with digits, `getUniqueName` will strip these digits and replace them with one larger than the maximum of the value of the stripped digits, and the digits on any name in `names` that has the same base name. So the Name 'AnotherTest1' gets the Label 'AnotherTest2'.

I suspect FreeCAD has been doing this for a long time, but recently the Model tree view has been changed to show the object Labels, and (in the case of Documents) not the Name, rather than the former behavious of just showing the Name.

I still don't understand why @maxwxyz observed the Label of the first document as being 'testFile1' though, as shown in the Model tree view.

kpemartin, 2025-01-20:
Further to the above, if there are no documents when `newDocument` is called (so the collection of existing labels is empty), the call to `getUniqueName` is not done and so a label that ends with digits is used unchanged.

kpemartin, 2025-01-20:
This might be less confusing if the Model tree view showed the Name for documents just as it does for objects within a document, and perhaps for both types, if the Name showed as a (read-only) property in the property list of the selected object.