FreeCAD/FreeCAD #19116: Start Page Preferences missing

reneas, 2025-01-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Maybe this is something that got removed with 1.0 but for me there is no possibility to customize the start page through Preferences. Neither in the list of settings nor in the workbenches settings:

[image]

Theoretically i only want to remove the examples from the Start Page but this is not possible. Even if I go into the parameter editor and set`Base app -> Preferences -> Mod -> Start -> ShowExamples -> false`. At first I thought this might be due to the fact that the Start Page is not a proper Workbench anymore but when I set the user parameter `ShowOnStartup` this actualy affects the behaviour.

### Full version info

```shell
OS: Nobara Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * btl 0.9.9
  * SearchBar 1.0.2
  * Manipulator 1.5.7
  * Curves 0.6.52
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-18:
It is not a workbench anymore

reneas, 2025-01-21:
@maxwxyz so it is not planned to be able to customize the layout of the startpage??

maxwxyz, 2025-01-21:
@reneas I have relabeled it, maybe you can add some more details how the start page should be customizable in the original issue

reneas, 2025-01-22:
so is there a possibility that this gets implemented?

luzpaz, 2025-01-28:
@reneas it may be better to break this ticket up into separate bugs. a Multi-bug ticket ends up either getting neglected or prematurely closed because they are hard to read and keep track of.

reneas, 2025-01-28:
@luzpaz thanks for your reply! 
I don't understand what you mean exactly, since I think this is a single feature request (be able to customize start page).