FreeCAD/FreeCAD #19121: TechDraw dimension numbers placed lower than normal

jlovett11211, 2025-01-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

WIth the latest update, all of my dimensions (ASME Referencing) are showing the numbers below center. The leader line now aligns with the acender of the measurement, not the center of the text. 

[image]

[p35.154 Model - HSE-T.Cross v12.FCStd.not.zip](https://github.com/user-attachments/files/18466091/p35.154.Model.-.HSE-T.Cross.v12.FCStd.not.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.67
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * freecad.gears 1.2.0
  * Gridfinity 0.5.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jlovett11211, 2025-01-18:
I just updated to the most recent version, and the issue is still there. 

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.67
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * freecad.gears 1.2.0
  * Gridfinity 0.5.0
  * Manipulator 1.5.7

Syres916, 2025-01-19:
It would appear https://github.com/FreeCAD/FreeCAD/commit/e9c30cf370d370e69683efd507ea982d6cdbf168 is not yet calculating the centre correctly and hence the text has been lowered. @WandererFan should basically the same code as `QuantitySpinBox` uses be relevant here https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/QuantitySpinBox.cpp#L804-L858 ?

WandererFan, 2025-01-19:
@syres916 - fontMetrics will tell you a lot about the font, but it won't tell you where your text gets painted within the bounding rectangle of the text item.   Widgets, on the other hand,  paint text where you tell them to paint it.

e9c30cf only fixed the painting outside the bounding rectangle.

WandererFan, 2025-01-19:
note: This seems to only affect ASME dims?  ISO dims look to be ok.

bguest, 2025-01-20:
I think the this PR (https://github.com/FreeCAD/FreeCAD/pull/19170) should fix the issue, but this will be my first PR on FreeCAD and the codebase is still pretty new to me so wish me luck.

luzpaz, 2025-01-28:
Testers needed to confirm fixes, any volunteers ?

luzpaz, 2025-01-28:
@jlovett11211 lets see, so the the fix for this went into the dev build revision 40007.  
We know that by checking out what commit was made in #19170  
We take the commit 999040803522e7b7ec3df6cc13c82baebe3f369d and run it through  

```shell
$ git rev-list --count 999040803522e7b7ec3df6cc13c82baebe3f369d                                                                                                                                                                                                                                                                           
40007
```
The revision is 40007. 
Now, lets see what the latest revision the dev builds[^1] were made from. 

[image]

Bummer, the previous dev builds is only at revision 40006 (one commit shy). Well we just need to wait a day or so for new builds to generate and the fix should be in there. Then you can download it and test.  

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

sean9721, 2025-02-01:
I can confirm the issue is gone in 40041.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/106.605
Installed mods: 
  * fasteners 0.5.36
  * sheetmetal 0.7.5
```

luzpaz, 2025-02-01:
Thanks for confirming. Closing.

closed 2025-02-01 by luzpaz