FreeCAD/FreeCAD #19122: unable to edit sketch 'invalid'

brettowe, 2025-01-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I did a trim edge on an arc after attaching another arc on it
afterwards while the sketch is valid enough to pad freecad keeps complaining the sketch is invalid 
the repair thing it keeps suggesting doesn't seem to have any issues with the sketch tho

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

None

### Anything else?

broken sketch attached
[broken-sketch.zip](https://github.com/user-attachments/files/18466526/broken-sketch.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-19:
[image]

brettowe, 2025-01-19:
i thought i had done that, anyhow doing it just deletes all constraints to 'fix' it
here is a pre broken sketch with some the constraints i had before just do the trim on the upper part where the arc is and it will break
I'll leave the bug open as i don't feel it should break like this with a simple trim
[pre-broken-sketch.zip](https://github.com/user-attachments/files/18470154/pre-broken-sketch.zip)

tarman3, 2025-01-19:
Did not see any problem

https://github.com/user-attachments/assets/2f7e8513-1517-4eac-90de-6fb3238b36e5

brettowe, 2025-01-21:
i don't see the malformed constraints message appear in solver messages
it goes from 11 DoFs to 30 DoFs and the constraints list goes empty

luzpaz, 2025-01-28:
Can we rename issue to succinctly describe what the actual issue is?

closed 2025-04-10 by hyarion