FreeCAD/FreeCAD #19123: [Performance] The software freezes for a few seconds when selecting a face.

luisDias89, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

How to replicate:
1º :Open the part.
2º: Select the Fasteners workbench.
3º Select a face.

From this moment on, even if we exit the workbench, the software continues to freeze.
PS: It is necessary to close FreeCAD to return to normal.

FILE:
[Extruder V0.zip](https://github.com/user-attachments/files/18467797/Extruder.V0.zip)

https://github.com/user-attachments/assets/3d964d50-3fa7-4b1f-9f05-0f73f2f774e5

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * ArchTextures
  * fasteners 0.5.34
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.9.1
  * Render 2024.12.15
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-19:
Seems like https://github.com/FreeCAD/FreeCAD/issues/18785 which should've been fixed by https://github.com/FreeCAD/FreeCAD/pull/19032?
@WandererFan @hyarion FYI

luisDias89, 2025-01-19:
@maxwxyz @WandererFan @hyarion I have been following the PR's, and I think this is something more specific. It only happens when the workbench fasteners are activated at least once. Version 1.0 also has this issue.

maxwxyz, 2025-01-19:
@shaise FYI

Syres916, 2025-01-19:
I think some context here might be useful, when #18785 was reported I could confirm that slowdown with my system but the alarming thing I noted while watching the video of this issue was the length of time Fasteners took to load, approx 16 seconds, that is definitely not normal. To compare, the following are mine:

_2.53 seconds_ for Fasteners to load with my day-to-day config files:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39900 (Git)
Build type: Release
Branch: main
Hash: 4c25c96d75effc6110cb03dc137df9e4abbd82d9
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * sheetmetal 0.7.3
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.35
```

_2.3 seconds_ for Fasteners to load with a clean config:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39900 (Git)
Build type: Release
Branch: main
Hash: 4c25c96d75effc6110cb03dc137df9e4abbd82d9
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * sheetmetal 0.7.3
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.35
```

Also I cannot replicate the issue so I would conclude there's something else in @luisDias89 config that's causing this particular issue. If the config files were renamed and not deleted and then the exact same steps repeated and report back the time taken to load Fasteners Wb it may help narrow down the root cause.

luisDias89, 2025-01-19:
@Syres916 The excessive time only occurs if switching workbenches with the face selected. Otherwise, it takes less than 1 second.

luisDias89, 2025-01-19:
Where and how do I do that?

Syres916, 2025-01-19:
> The excessive time only occurs if switching workbenches with the face selected. Otherwise, it takes less than 1 second.

On my system, absolutely no difference whether a face is selected or not.

> Where and how do I do that?

To rename your config files on a Windows system, navigate to (the default location of) `C:\Users\`_username_`\AppData\Roaming\FreeCAD` and rename the two files `user.cfg` & `system.cfg` to something like `user_old.cfg` & `system_old.cfg`
When you start FreeCAD it will create new ones automatically as if you are a new user but all your Addons will still be available.

JohnOCFII, 2025-01-19:
> To rename your config files on a Windows system, navigate to (the default location of) `C:\Users\`_username_`\AppData\Roaming\FreeCAD` and rename the two files `user.cfg` & `system.cfg` to something [...]

You could also use the new "Restart in Safe Mode" from the `Help` menu to achieve the same effect with less manual effort.

Syres916, 2025-01-19:
> You could also use the new "Restart in Safe Mode" from the `Help` menu to achieve the same effect with less manual effort.

And how would you get to Fasteners Wb when Safe Mode disables all Addons as well as running clean config files???

hyarion, 2025-01-19:
> > You could also use the new "Restart in Safe Mode" from the `Help` menu to achieve the same effect with less manual effort.
> 
> And how would you get to Fasteners Wb when Safe Mode disables all Addons as well as running clean config files???

This is indeed an issue. It would be really nice if it was possible to restart into the same safe mode.

JohnOCFII, 2025-01-19:
> And how would you get to Fasteners Wb when Safe Mode disables all Addons as well as running clean config files???

I don't know.  I forgot that Safe Mode also disables all Add-ons.

JohnOCFII, 2025-01-19:
> This is indeed an issue. It would be really nice if it was possible to restart into the same safe mode.

I wonder if Safe Mode could have an option to just run with clean config files and leave the add-ons activated?

luisDias89, 2025-01-20:
@Syres916 I have already renamed user_old.cfg and system_old.cfg, opened FreeCAD without any configuration, and got the same result. It was blocking the selection of the face. 
After that, I removed all add-ons from FreeCAD and encountered the same problem.

https://github.com/user-attachments/assets/2e2ea5d5-6b90-42b8-944b-ef6b81d4a002

luisDias89, 2025-01-20:
I even switched to Linux to try it out, and it's the same. Are you using this model I sent to test on your computers?

https://github.com/user-attachments/assets/fdda28d6-8933-4f92-b03f-4c79ae84edbe

luisDias89, 2025-01-20:
@Syres916 And it can't be just any face, only 3 or 4 faces of this model cause the issue. And only after entering the Fasteners Workbench, which is even stranger...

https://github.com/user-attachments/assets/6b6d6a84-2277-47b4-92d0-b9fe7e6a3064

shaise, 2025-01-20:
I can confirm I see the same issue. Let me investigate if it is a Fasteners issue.

Syres916, 2025-01-20:
> Ok. It is definitely a Fastener issue. I'm on it

Specifically with the built-in Assembly Wb?

shaise, 2025-01-20:
No, Just an extremely complex algorithm to decide if the Move command is active or not.
It is fixed now. Fasteners should be updated

luisDias89, 2025-01-20:
@shaise Perfect, its fixed!! Ty

closed 2025-01-20 by luisDias89

maxwxyz, 2025-01-20:
Closing this when this is an issue in the addon