FreeCAD/FreeCAD #19124: Sketcher: Offset is not applied when entering a negative value

Pendelf, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Offset is not applied when entering a negative value
Select geometry, click on offset, press J for constraint, enter a negative value, press Enter and nothing happens

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.53
  * fasteners 0.5.35
  * PieMenu 1.11.2
  * Road 2025.1.1
  * sheetmetal 0.7.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-19:
I guess it it not possible to know which side is the negative one @PaddleStroke 
@Pendelf does it work when moving the mouse to the "negative" side and setting a positive value?

Pendelf, 2025-01-19:
@maxwxyz 
Works if you do an offset inward without entering a value. The value can then be changed to the desired value. The value is without a minus sign

maxwxyz, 2025-01-19:
Seems like an intended behavior. Similar as you can dimension a line between an axis with a value, it is not clear to which side it is dimensioned.

Pendelf, 2025-01-19:
I think the ability to offset inwards should be there.

https://github.com/user-attachments/assets/894ae8e2-f304-4b97-8365-4bc6e56ae07e

PaddleStroke, 2025-01-19:
I see. Yes you're right. It should be possible to enter a negative value.

Pendelf, 2025-01-25:
By the way, in version 1.0 the offset works both ways

PaddleStroke, 2025-01-25:
Ah so it's a regression. I had the feeling that this was working before indeed.

coralof, 2025-01-25:
I may have the same issue? When trying to create a negative-value offset geometry, and entering the number in the field and pressing enter, FreeCAD completely freezes and has to be force-quit.

If I just place the offset without typing in the number by scaling it to where I want and pressing enter, it works fine. I've closed and reopened FreeCAD a ton of times, and it always crashes when entering the negative value manually in the field:

[image]

My own system info:

OS: Fedora 41
FreeCAD:

[image]

luzpaz, 2025-01-26:
What code caused the 'regression' ?

Pendelf, 2025-01-29:
@furgo16 

https://github.com/user-attachments/assets/e2ea431d-2c8a-4f5b-b09e-5ea2f17a7d72

furgo16, 2025-01-29:
It does not seem to appear on the video (the dialog seems to be off screen), but I understand that the workaround is to first place the offset without typing, and then change the constraing by typing a _positive_ offset. That worked for me. Thanks!

PaddleStroke, 2025-02-21:
@ziad-ashraf7 if you are interested in another small bug that should not be too hard to fix, there's this regression that is a good candidate.

furgo16, 2025-02-21:
Also note that the workaround is not perfect: I can't recall the detail since it's been a while since I used it. But after having applied a positive offset in place of a negative one, then there is an error when modifying the sketch.

The workaround that I followed after I noticed this was to use 1.0.0 whenever a negative offset was needed, and only then switch to 1.1.0dev.

maxwxyz, 2025-03-04:
This is still an issue

Pendelf, 2025-03-04:
@maxwxyz 
Yes, in the last build, everything is as it was

ziad-ashraf7, 2025-03-05:
> [@ziad-ashraf7](https://github.com/ziad-ashraf7) if you are interested in another small bug that should not be too hard to fix, there's this regression that is a good candidate.

Sorry, I just saw your mention. I think I need to recheck my notification settings !
I think I can give this issue a look to solve it as a preparation for GSoC 2025.
Thank you for your mention, and sorry for late response!

ziad-ashraf7, 2025-03-09:
I've made some log messages in `DrawSketcherHandlerOffset.h` file to track the Offset process.
and when I set a positive value to the SpinBox, I can see the logs, but when entering a negative value, I can't see them and the negative value is not set.

so I tried to add some logs on other files, like SketcherToolDefaultWidget.cpp / .h.
and again. i can see the logs only when entering a positive value but not a negative value.

My intuition:
I think that there is a part on the code that sets the min value of the SpinBox in the UI component that prevents the negative value before passing it to the logic (where the lgos are written).
that's the reason why I can't set the value to be negative , and the logs is not appears.

@PaddleStroke may I ask you for some help in this problem? is my intuition is correct ? Where should I find the piece of code that is responsible for this , I have edited the `QuantitySpinBox::setMinimum()` function and nothing happened!

PaddleStroke, 2025-03-10:
I think you are on the right track. looking for setMinimum in the code finds configureParameterMin in SketcherToolDefaultWidget.cpp but that does not seem to be used by the offset DSH. So it can't be this.

One thing you can do is check the history of the offset DSH and try to replace the current version of the DSH by an older version to see if it works again. If not then you can rule out the offset DSH. Then the change must be outside. And yes it sounds likely something set a minimum to the spinbox.

In this case it can be interesting to test with another tool to see if you can type in a negative value. For instance rectangle with frame option. Or any tool with an angle like slot or arc.

PaddleStroke, 2025-03-10:
I have done these tests and confirm all tools cannot accept negative inputs. So it must be something about setMinimum indeed.

PaddleStroke, 2025-03-10:
After further test the problem is not in sketcher altogether. It's in core.
`EditableDatumLabel::valueChanged` signal does not trigger if the value is negative.
So `QuantitySpinBox::valueChanged` does not trigger either.

PaddleStroke, 2025-03-10:
I'm betting without testing yet that the problem comes from https://github.com/FreeCAD/FreeCAD/commit/b470f74398945b1549617929f6275cdd13420de5 @kadet1090

PaddleStroke, 2025-03-10:
Just tested and confirm. 

```
    if (isEnter && !isNormalized()) {
        normalize();
        return;
    }
```

What is that ? No spinbox accept negative input accross the whole application. Please consider fixing this

kadet1090, 2025-03-10:
> Just tested and confirm. 
> 
> ```
>     if (isEnter && !isNormalized()) {
>         normalize();
>         return;
>     }
> ```
> 
> What is that ? Currently no spinbox can accept negative input accross the whole application. Please consider fixing this

I am almost sure that spinboxes can accept the negative inputs. If anything the isNormalized method should be fixed.

This was added because the spinbox supports simple math but there is no way to see the result of calculation. In all software it works like this - you can enter formulas into numeric inputs and it is normalised into number when you blur the input or when you hit enter. Once you see the number you should be able to hit enter again to reconfirm. 

The problem here probably is because for some reason negative number is not treated as normalized and that would be the `isNormalized` method - and it should be fixed there.

PaddleStroke, 2025-03-10:
You are correct my above statement was not correct. It seems to happen only for OVP.

ziad-ashraf7, 2025-03-10:
> I am almost sure that spinboxes can accept the negative inputs. If anything the isNormalized method should be fixed.
>
> This was added because the spinbox supports simple math but there is no way to [...]

I was debuging on this file and this statment , the input from the user 'validStr' is not as same as the 'd->quantity' , i'm still working on this issue and i this i'm very close to find what should i do about it.

ziad-ashraf7, 2025-03-10:
> Just tested and confirm.
> 
> ```
>     if (isEnter && !isNormalized()) {
>         normalize();
>         return;
>     }
> ```
> 
> What is that ? Currently no spinbox can accept negative input accross the whole application. Please consider fixing this

Yes you are right, also me thinks that the problem might be here , but if you tried to debug and entering a negative value , you might see the user input in 'validStr' and whan you want to see the 'd->quantity' you won't see the same value of what you entered.

PaddleStroke, 2025-03-10:
@ziad-ashraf7 check my comment here https://github.com/FreeCAD/FreeCAD/pull/20101#issuecomment-2710130578

I think the EditableDatumLabels are not compatible with the new system Kadet has introduced because they are not following keyboard keys and instead rely on the enter key to update the value

PaddleStroke, 2025-04-01:
@kadet1090 bump if you can check this bug out.

closed 2025-05-12 by chennes