FreeCAD/FreeCAD #19125: The new transformation is slowing down

Pendelf, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In version 1.0 with the old transformation there are no brakes, but the new transformation slows down when you move it

https://github.com/user-attachments/assets/9c0080a6-4c66-4fa9-aad8-80e69ee87486

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.53
  * fasteners 0.5.35
  * PieMenu 1.11.2
  * Road 2025.1.1
  * sheetmetal 0.7.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-19:
@kadet1090 FYI

tarman3, 2025-01-19:
Can confirm in Linux

https://github.com/user-attachments/assets/599d239c-9fef-44c3-9ac3-48e828d85156

<details>
  <summary>FreeCAD 1.1.0dev.39841 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.35; Quetzal.backup1736851251.7081769 (Disabled); sheetmetal 0.7.3; btl 0.9.9; Silk 0.1.5; Curves 0.6.53; Defeaturing 1.2.3
```

</details>

CPU: dual core Intel Core i3-6006U (-MT MCP-)
Kernel: 6.12.9-arch1-1 x86_64
Mem:4.06/7.63 GiB

Pendelf, 2025-01-19:
Tomorrow I'll try it on Linux, maybe it's Windows

ovo-Tim, 2025-01-21:
Might relate to #19128

kadet1090, 2025-01-21:
I am not able to reproduce it on Linux, doing profiling also does not show anything. I need some profiling data from windows to understand what is going on.

Pendelf, 2025-01-22:
@kadet1090 
Specifically what information do you need and where to get it?

kadet1090, 2025-01-22:
@Pendelf flamegraph limited to when transform tool is used would be useful or cachegrind file of such case.

Pendelf, 2025-01-24:
Today it was possible to try on Windows 10 and here everything works well without brakes

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

Pendelf, 2025-01-26:
Today I tried to install FreeCAD 1.1 in Windwos Sandbox on the same machine on which the slowdowns are present and in Sandbox everything is normal, there are no slowdowns. I think the problem can be closed, most likely the matter is in my settings or in one of the addons.

I also tried to run FreeCAD in safe mode and the slowdowns did not disappear

maxwxyz, 2025-01-26:
Thanks for testing

closed 2025-01-26 by maxwxyz