FreeCAD/FreeCAD #19126: Sketch on surface -> thickness not working anymore

MarcEngrie, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

While executing the sample/exercise shown by MangoJelly (https://www.youtube.com/watch?v=th4eQmskTnc) an acquaintance and I came across the problem that when applying the thickness (+ or - ) on a skecth_on_surface as shown around 21' 30" in the video, doing this with the released version of FreeCAD 1.0 as well as with the weekly build will generate an error (see below) and the thickness will not be applied. 

"[15:15:05](https://www.youtube.com/watch?v=th4eQmskTnc&t=54905s)    0:Invalid final face  1:Invalid final face  2:Invalid final face  4:Invalid final face  5:Invalid final face  6:Invalid final face 14:Invalid final face 17:Invalid final face 21:Invalid final face 25:Invalid final face 28:Invalid final face 29:Invalid final face 30:Invalid final face 32:Invalid final face 33:Invalid final face 34:Invalid final face 36:Invalid final face 37:Invalid final face 38:Invalid final face 40:Invalid final face 41:Invalid final face 42:Invalid final face 44:Invalid final face 45:Invalid final face 46:Invalid final face<Part> ViewProviderExt.cpp(1308): Cannot compute Inventor representation for the shape of Unnamed[#Sketch_On_Surface](https://www.youtube.com/hashtag/sketch_on_surface): Geom2d_TrimmedCurve::U1 == U2

[15:17:36](https://www.youtube.com/watch?v=th4eQmskTnc&t=55056s)    0:Invalid final face  1:Invalid final face  2:Invalid final face  4:Invalid final face  5:Invalid final face  6:Invalid final face 10:Invalid final face 14:Invalid final face 18:Invalid final face 21:Invalid final face 22:Invalid final face 25:Invalid final face 26:Invalid final face 28:Invalid final face 29:Invalid final face 30:Invalid final face 32:Invalid final face 33:Invalid final face 34:Invalid final face 36:Invalid final face 37:Invalid final face 38:Invalid final face 40:Invalid final face 41:Invalid final face 42:Invalid final face 44:Invalid final face 45:Invalid final face 46:Invalid final face<Part> ViewProviderExt.cpp(1308): Cannot compute Inventor representation for the shape of Unnamed[#Sketch_On_Surface](https://www.youtube.com/hashtag/sketch_on_surface): Geom2d_TrimmedCurve::U1 == U2"

Tested on FreeCAD 1.0 (released version

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Belgium (en_BE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.33
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears 1.3.0
  * LCInterlocking
  * sheetmetal 0.6.1
  * woodworking 0.21.2.33771.0.23.1

Tested on FreeCAD 1.1.0 Dev weekly build

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Belgium (en_BE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.33
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears 1.3.0
  * LCInterlocking
  * sheetmetal 0.6.1
  * woodworking 0.21.2.33771.0.23.1

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Belgium (en_BE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.33
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears 1.3.0
  * LCInterlocking
  * sheetmetal 0.6.1
  * woodworking 0.21.2.33771.0.23.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-19:
Please attach file

MarcEngrie, 2025-01-19:
> Please attach file

Sorry. Attached now but since it seems not allowed to attach .FCstd file I renamed it to .txt. So file is added in original post now.

tarman3, 2025-01-19:
https://streamable.com/z012u4

Note: Part with dark red faces is flipped

[image]

maxwxyz, 2025-01-19:
Did this fix the issue? Just comment if not and we can reopen

closed 2025-01-19 by maxwxyz

MarcEngrie, 2025-01-20:
If only using the hexagon-sketch itself, it works fine. However is using the arrary based on the hexagon-sketch, thats were the error pops up and FreeCAd fails to execute. It seems to have worked in 0.21/0.22 (cfr video of MangoJelly) but not anymore in 1.0.

MarcEngrie, 2025-01-20:
@maxwxyz  No, didnot fiw issue. It seems, to be more specific, "Draft -> array" in combination with "sketch on surface" related.

maxwxyz, 2025-01-20:
You should open an issue here: https://github.com/tomate44/CurvesWB

MarcEngrie, 2025-01-21:
Thx. Done!