FreeCAD/FreeCAD #19128: Recompute Part Design objects time gradually increase.

ovo-Tim, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi, while I was trying to measure the recomputing time, I found the recomputing time of Part Design objects gradually increased. So, I wrote a simple script to determine:
``` python
import random, timeit
obj = FreeCAD.getDocument('Unnamed1').getObject('Pad')
def test():
	obj.Length = f'{random.randint(5,15)} mm'
	obj.recompute()

print("Test start!")
for i in range(8):
	print(i, ':', timeit.timeit(test, number=100))
```
[image]
And the result is here:
```
Test start!
0 : 0.7128433370016865
1 : 1.442012331001024
2 : 2.167117896999116
3 : 2.9721044629986864
4 : 3.8881915220008523
5 : 4.778530065999803
6 : 5.679119818996696
7 : 6.794938696002646
```

I also tested Extrude created by Part workbench and it don't have such problem.

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.0, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Render 2024.12.15
  * Assembly4 0.50.15
  * A2plus 0.4.68
  * EasyProfileFrame
  * frameforge 0.1.0
  * CfdOF 1.29.2
  * MakerWorkbench 1.0.1
```

### Subproject(s) affected?

PartDesign

### Anything else?

I'm willing to contribute, but I only know a little about C++. If there is anything I can help, feel free to ask me. 🤗 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-19:
Is it too low or what is the actionable issue?
@kadet1090 maybe for part/part design merge?

ovo-Tim, 2025-01-20:
> Is it too low or what is the actionable issue?

It might not cause a significant issue in the short term, but since FreeCAD will gradually become slower and slower, I think there is a need for repair. It may reveal a memory leak issue or something like that.

luzpaz, 2025-01-20:
It would be great if someone profiled the operation to discover if there was a memory leak.

ovo-Tim, 2025-01-20:
@luzpaz Although I only know a little about C++, I still build [this](https://github.com/ovo-Tim/ASan-FreeCAD-builder) to analysis the memory leak.
And here is the result after I ran my script:
[log.log](https://github.com/user-attachments/files/18479880/log.log)

ovo-Tim, 2025-01-20:
I also drew this wordcloud, hope can help something:

[image]

ovo-Tim, 2025-01-21:
It looks `Materials.so` has memory leak problem seriously.

[image]

maxwxyz, 2025-01-21:
@ovo-Tim could you please try on the latest weekly, there was a materials fix recently

ovo-Tim, 2025-01-21:
> @ovo-Tim could you please try on the latest weekly, there was a materials fix recently

[This FreeCAD with ASan](https://github.com/ovo-Tim/ASan-FreeCAD-builder) was built from main branch yesterday. 

But I don't know if ASan influenced the test result, so I will test weekly version later.

ovo-Tim, 2025-01-21:
@maxwxyz Things didn't change at all.
```
./FreeCAD_weekly-builds-39896-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.1.0, Libs: 1.1.0devR39896 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
Test start!
0 : 0.39974118899999667
1 : 0.7318726129999504
2 : 1.0801984889999972
3 : 1.479454091999969
4 : 1.8920787999999789
5 : 2.244531980999966
6 : 3.29820737
7 : 3.8444893860000207
```

maxwxyz, 2025-01-21:
@davesrocketshop seems to be in material, could you please take a look?

davesrocketshop, 2025-01-21:
> It looks `Materials.so` has memory leak problem seriously.
> 
> [image]

This may not be an issue. The materials are loaded when first referenced and then retained. I'll dig into it though... it should have no influence on recompute times.

FEA-eng, 2025-12-14:
> [@FEA-eng](https://github.com/FEA-eng) can you explain the results ?

Those are recompute times from tests created by the OP, and from what I understand, it's bad that they are increasing.

pieterhijma, 2025-12-23:
Great that there is progress on this issue. So, it is clear that it is OCC related, but is it a problem in OCC I should act on?

Lgt2x, 2026-01-02:
Fixed in the official weekly builds by https://github.com/FreeCAD/FreeCAD/pull/26363 for me , can you confirm so we can close it @kadet1090 ?

closed 2026-01-02 by maxwxyz