FreeCAD/FreeCAD #19129: In FC Light and Dark theme, Undo-Redo icons and their small arrows overlap each other

ErtugrulEge, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello, I think there is a small user interface problem. Light and Dark theme's  undo-redo icons overlap with their small arrows. FC Classic is working well.I think, Light and Dark themes should behave  like Classic theme's undo-redo icons.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-01-19:
Suggestion given on an existing Pull Request https://github.com/FreeCAD/FreeCAD/pull/18929#issuecomment-2600936430

maxwxyz, 2025-01-19:
I thought there was already an issue about that but I cannot find it.
@MisterMakerNL FYI

MisterMakerNL, 2025-01-19:
It's already fixedin https://github.com/FreeCAD/FreeCAD/pull/18929
And yes it's an old qt/Freecad bug.

closed 2025-01-27 by yorikvanhavre