FreeCAD/FreeCAD #19131: CAM pocket operation fails on OpenSCAD model depending on cube length

bzuu, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I imported an OpenSCAD model and try to generate a pocket toolpath in the CAM workbench. This operation fails on the specific model, but works if the dimensions in the imported geometry are different.

The attached FreeCAD model contains a very simple imported OpenSCAD model consisting of a difference between two cubes. The pocket operation in CAM does not generate a path for this particular model. When however I reduce the length property of the 'cube' component from 400mm to 300mm it all works fine.

Generated path in original model
[image]

Generated path in model with reduced length
[image]

Attached are:
* [The FreeCAD model](https://github.com/user-attachments/files/18469460/openscad_cam_test.zip)
* [The original OpenSCAD file](https://github.com/user-attachments/files/18469461/cam_import_test.zip)

### Full version info

```shell
Main installation:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Netherlands (en_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.6.0

Also tried in latest development build:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Netherlands (en_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.6.0
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-19:
Pocket feature contain a lots of bugs
Try change depth values little bit

I realize it's like a circus ))

[image]

https://github.com/user-attachments/assets/6d25080c-5936-44b9-8aab-3c0af63d8237

tarman3, 2025-01-19:
Add parameters BooleanFuzzy
Top Menu -> Tools -> Edit Parameters -> BaseApp -> Preferences -> Mod -> Part -> Boolean 

New float item
BooleanFuzzy = 1

Close FreeCAD
Run FreeCAD
Open Pocket operation
Change any value and press *Apply**

bzuu, 2025-01-19:
Thanks @tarman3 that seems to work indeed. I have no idea what that parameter does however, is this something I as a new user should have been able to figure out and does this mean the issue I reported is not a bug?

tarman3, 2025-01-19:
This is known problem and reported many times

closed 2025-01-19 by maxwxyz