FreeCAD/FreeCAD #19134: Unit test Document.DocumentObserverCases.testDocument fails (for me) using Gui (non-cmd) FreeCAD

kpemartin, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The version of FreeCAD that I build from source fails the `Document.DocumentObserverCases.testDocument` unit test.
The failure is because the document observer receives an unexpected notification.

I'm building the Release build on Windows 10 using Visual Studio 2022 17.11.2 and libpack 1.0.0-v3.0.0-Release and also running the app on Windows 10.

This test case is in src\Mod\Test\Document.py starting at line 2111.
After closing a document with the Gui up, the code expects to have received a `DocDeleted` notification on the closed document, and conditionally a `DocActivated` notification as the previous document becomes the active document again.

The test *only* expects the second notification if `not FreeCAD.Gui.HasQtBug_129596`. This property `HasQtBug_129596` is true on my builds yet the notification takes place, causing the test to fail. Code in `Application.cpp` sets this property True iff `#ifdef HAS_QTBUG_129596`

This define is optionally set in Application.h:
```
#if QT_VERSION >= QT_VERSION_CHECK(6,0,0) && QT_VERSION < QT_VERSION_CHECK(6,8,1)
# define HAS_QTBUG_129596
#endif
```
and apparently my QT version is 6.7.2 so it is indeed being defined.

The definition of `HAS_QTBUG_129596` also stops MainWindow.cpp from executing:
```
d->mdiArea->setActivationOrder(QMdiArea::ActivationHistoryOrder);
```

This particular QT issue is documented [here](https://bugreports.qt.io/browse/QTBUG-129596)

This change was done as commit cdb1a4f which unfortunately does not explain why the change removes the expected notification (nor for that matter why it is an effective workaround for the issue). Perhaps with the change, *no* MDI window is activated, which would likely explain why it fixed the bug and also why the unit test no longer expects a notification.

It would however appear that, at least in my environment, another MDI window gets activated, so the notification is being generated.

This makes me wonder if the bug this is trying to fix still occurs in my environment. I will look at the directions and see if I can reproduce it. It is possible, though, that the QT implementation for Windows is sufficiently different that the bug does not (and never did) occur.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39863 (Git)
Build type: Release
Branch: Issue18679
Hash: ec61245fff58c2d973e3e493e10264cf58e2e9ab
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-19:
@pinkavaj FYI

kpemartin, 2025-01-20:
Note that once the test fails, it leaves a document called "Observer1" open, and subsequent runs of the test succeed unless this document is closed first.

In fact, the test passes if any document is already open when the test is run.

kpemartin, 2025-03-31:
I have not been noticing this lately. Perhaps my change to libpack 1.1 has corrected the problem

closed 2025-09-26 by kpemartin