FreeCAD/FreeCAD #19138: IFC export with transparent doors and window frames

semhustej, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When exporting attached file to IFC format, doors and window frames are transparent. I am able to reproduce the issue only with 1.1dev version. 1.0 exports doors and windows well.

The exported IFC file was tested in two IFC viewers - BIMvision and BonsaiBIM.

Export from 1.1:

[image]

Export from 1.0:

[image]

The file, including IFC export: 

[FC-transaprent-doors.zip](https://github.com/user-attachments/files/18469990/FC-transaprent-doors.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

Might be related to https://github.com/FreeCAD/FreeCAD/issues/18655  - this one is not about IFC export though.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-19:
@yorikvanhavre is this the same as #18655 ?
Does either https://github.com/FreeCAD/FreeCAD/pull/18761 or https://github.com/FreeCAD/FreeCAD/pull/18767 solve this issue?

semhustej, 2025-01-20:
To clarify: the doors and windows are OK in FreeCAD (unlike with https://github.com/FreeCAD/FreeCAD/issues/18655) . The problem only occurs in IFC export.

I was also not able to reproduce the issue with simple file i created (wall, window, door). So far it occurred to me only in the OP file.

yorikvanhavre, 2025-01-20:
The internal colors have switched from rgbt to rgba. There must be some code in the IFC export that must be adapted. I'll have a look

tetektoza, 2025-05-11:
@semhustej Got a solution for this one already, but tell me, I will ask it from the other side. How do you set transparency of an object? Like let's say I want it to be transparent?

semhustej, 2025-05-12:
@tetektoza 

As far as I know, there are three options how to set transparency of an object. At the end, I think all of them just change the View Properties -> Shape Appearance -> Transparency value:

1. View Properties -> `Transparency` parameter
2. Using BIM Materials. It is possible to create BIM materials through BIM_Material tool. One of BIM material properties is `Transaprency`. When BIM material is assigned to objects, the View Properties -> `Transparency` parameter is set to the one that the BIM material uses. Windows use this material system.
I did a test and it seems when exiting BIM Material transparency is changed, it is not propagated to objects having this material. For instance color gets propagated. I guess I will report this as a separate issue.
3. Using Materials from the Material workbench. This is a new option since FC 1.0, it is meant to unify the use of materials throughout all FreeCAD workbenches. It has not yet been unified with BIM materials yet, I don't know what the timeline or roadmap is. But one can assign these global materials to BIM objects no problem (by selecting object, right-clicking and selecting `Material` or `Appearance` from the menu). These can have transparency set too.

Windows have to use BIM Multi-material to have different material assigned to the window pane and the window frame

I recorded a short video showcasing these three options

https://github.com/user-attachments/assets/37f929b9-a903-4f56-b222-afd4ce347d98

 
Don't hesitate to ask more questions. I will be happy to help if I can :)

furgo16, 2025-05-17:
@Roy-043 's [findings on the forum thread](https://forum.freecad.org/viewtopic.php?p=827133#p827133) might be relevant here too:

> The transparency issue may be due to a mix-up with the alpha channel.
>
>I think we need to look at the `IFCSURFACESTYLERENDERING` values in the IFC file, and in `exportIFC.py` the lines with `createIfcPresentationStyleAssignment` are relevant.
>
>In `Ifc-Substractions-v2.ifc` changing:
>`#209=IFCSURFACESTYLERENDERING(#208,1.,$,$,$,$,$,$,.FLAT.);`
>to:
>`#209=IFCSURFACESTYLERENDERING(#208,0.,$,$,$,$,$,$,.FLAT.);`
>Fixes the transparency of the door.

semhustej, 2025-05-17:
@furgo16 I think it's already mostly fixed with this PR: https://github.com/FreeCAD/FreeCAD/pull/21289

closed 2025-05-23 by hyarion