FreeCAD/FreeCAD #19141: hard to select correct transform elements(arrow/ball) at certain angles

kinghat, 2025-01-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

it is hard to select the proper transform element, either the arrows or the balls, when youre at an unfavorable angle. you end up spinning the model out of place:

[image]

### Full version info

```shell
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.3
  * fasteners 0.5.35
  * ThreadProfile 1.92.0
  * Gridfinity 0.8.0
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

this is an issue anywhere these elements are used. hopefully a solution would be to highlight which elements youre hovering over.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-23:
@kinghat any proposal to make this more friendly ?

cc @kadet1090

kinghat, 2025-01-23:
> [@kinghat](https://github.com/kinghat) any proposal to make this more friendly ?
> 
> cc [@kadet1090](https://github.com/kadet1090)

the solution would be to highlight the "ball" and arrow dragging points when hovered so you know where youre at.