FreeCAD/FreeCAD #19147: Problem with walls using BIM multi-material

Freakcad-User, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Multimaterial wall length is incorrect, and grows (seemingly exponentially) longer each time a recompute occurs.

### Full version info

```shell
Discussed on the forum here:
https://forum.freecad.org/viewtopic.php?t=93610&sid=3c23e4c1ddcd88055a7fca5e5ec66e4f

I'm encountering a problem with walls that use a wall preset based on a multi-material. Here's what's happening:

I have already created the multimaterial which is 3 layers (stud wall with gypsum board on each side) that creation process works fine.

Now I create a wall. When I click the second point the wall that is created is longer than what I specified.

In properties the wall says one length, but measures (with the measure tool) a different length that what the properties panel shows. The measure tool seems to be showing the correct length.

Now the really strange part--when I add anything else to the drawing (walls, lines, doors) then any walls of that preset multi-material type that I have created before grow in length... a lot, they all keep growing each time I add something to the drawing.

It seems to happen when a recompute occurs, but it keeps growing every time it recomputes. Also, I notice that normally the refresh icon goes gray right after a recompute, but when the multi-layer wall is present it doesn't ever go gray, and the wall grows in length (seemingly at an exponential rate) each time I hit refresh.

I am not having a problem with "generic" non-multimaterial walls.

paullee asked if it might be related to:
#19083 "this is a bug in FreeCAD" message when changing wall length

My issue occured in version 1.0 (stable)
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * PieMenu 1.7.6
  * OpenTheme 2024.9.1

Then went away on version:
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * PieMenu 1.7.6
  * OpenTheme 2024.9.1

Now has returned on this build:
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * PieMenu 1.7.6
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Anything else?

This is my first bug report. I hope I did it correctly!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-20:
@yorikvanhavre FYI, will this also be fixed when using the new Materials?

semhustej, 2025-01-20:
The discussed file: 

[2025-01-05_4411-Wilson_Bath-1-Reno.FCStd.zip](https://github.com/user-attachments/files/18481406/2025-01-05_4411-Wilson_Bath-1-Reno.FCStd.zip)

Demonstration of the behavior:

https://github.com/user-attachments/assets/c42cec68-53a1-49d1-9f57-17a2ef25508d

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

Freakcad-User, 2025-01-20:
Thanks for adding the video.  This is exactly what I experience with the issue.

paullee0, 2025-01-23:
OK, thanks, I see where it goes wrong and the concept of length of a wall would be confused by multi-material.  

Would see what best to be done soon :)

paullee0, 2025-01-24:
OK, prepared a fix, posted at FC forum and here now.

Anyone can help testing?  Thanks.

[ArchWall.py.txt](https://github.com/user-attachments/files/18538661/ArchWall.py.txt)

semhustej, 2025-01-24:
> OK, prepared a fix, posted at FC forum and here now.
> 
> Anyone can help testing? Thanks.
> 
> [ArchWall.py.txt](https://github.com/user-attachments/files/18538661/ArchWall.py.txt)

I quickly tested the fixed library.

- the OP file requires refresh only once, the length of the wall does not "grow" after that.
- tried placing door in other wall in OP file. Everything works fine.
- tried creating a new wall from a line and a sketch. Applied said multi-material to those. Seems to work OK.

Thank you for the fix!

paullee0, 2025-01-25:
In fact, this bug only affect Base object with only 1 edge and Multi-material is used.

closed 2025-01-27 by yorikvanhavre