FreeCAD/FreeCAD #19150: Null input shape when applyin thickness after a sweep

vicholp, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi!

I'm triying to apply a thickness skin in the Part Workbench to the `Sweep001` object, made from two different faces and a simple line:

[image]

But getting this exception:

```
<Exception> TopoShapeExpansion.cpp(2980): Null input shape
Thickness: Null input shape
```

In the same project I have another similar object `Sweep`, made from two equal faces, and a more complex line:

[image]

And the thickness works without problem.

I have attached the project file to this issue, and a video reproducing the last steps for the error. 

Note that with one shape it works and with the other one not. Also, note that applying thickness to the `Fusion` also does not works.

Thanks in advance.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CfdOF 1.29.4
  * Render 2024.12.15
```

### Subproject(s) affected?

Part

### Anything else?

[project.zip](https://github.com/user-attachments/files/18472375/project.zip)

https://github.com/user-attachments/assets/c76ecefc-9b25-4c32-a84b-0c26c59767c8

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-11-03:
I get:

`Thickness: BRep_API: command not done`

But also some TNP error when recomputing:

`11:24:35  <Sketch> SketchObject.cpp(9092): External geometry example#Sketch.e27 missing reference: Sketch001.;g10v2;SKT;:H,V`

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251102 (Git shallow)
Build date: 2025/11/02 16:03:37
Build type: Release
Branch: main
Hash: 45ca23d074f39a9b586c8320f1e7cd20ca466fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

ElementW, 2026-08-25:
Still relevant as of faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and either OCC 7.9.3 or 8.1.0.dev1-3d097a0328